kwindrem/VeCanSetup #4: CANable fails to initialise on RPI reboot.  Disconnect / reconnect seems to get it working again.

andyalford, 2022-06-20:
I have just upgraded 2x Raspberry Pis, one 3B+ with 2.90-14 Pi2, and one 3B with 2.90-14 Pi2 Large.
I have a Canable Pro which has been working for a few years with previous release and the manual install
I have used SetupHelper and all working except this problem with VECanSetup.

I have REC-BMS enabled using Settings/Services/Can0 = Can.bus 250kbps
In SSH, I selected 'a' and chose '2' for slcand 

Problem Symptoms:  
	1. It works for a while, but when I restart the Pi, the REC-BMS disappears.  
	2. A fix appears to be to disconnect the USB cable and reinsert it, and the REC-BMS appears in the device list and works.
	3. I'm getting the same behaviour on both Pis, so that elimates hardware issues I hope.

So it seems to be some kind of initialisation problem on restart that is locking up the comms???

Thanks for your work, this will really help out a lot of people.

Andy Alford, mail@andyalford.com

andyalford, 2022-06-20:
{SYSTEM WORKING AFTER DISCONNECT/RECONNECT USB for CANABLE}

root@raspberrypi2:/usr/local/bin# lsusb
Bus 001 Device 011: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 012: ID ad50:60c4
Bus 001 Device 008: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 006: ID 0781:5590 SanDisk Corp. Ultra Dual
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 007: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 005: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 009: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@raspberrypi2:/usr/local/bin#
The system is going down for reboot NOW! (console) (Mon Jun 20 06:27:46 2022):
client_loop: send disconnect: Connection reset

{VENUS STARTS UP AGAIN, BUT REC-BMS NOT SHOWN IN THE DEVICE LIST}

C:\Users\Andy>ssh root@192.168.1.199
root@192.168.1.199's password:
Last login: Mon Jun 20 06:27:59 2022
root@raspberrypi2:~# lsusb
Bus 001 Device 011: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 010: ID ad50:60c4
Bus 001 Device 008: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 006: ID 0781:5590 SanDisk Corp. Ultra Dual
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 007: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 005: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 009: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@raspberrypi2:~# dmesg | grep -i can
[    6.774984] usb 1-1.2.3: Product: CANable 1205aa6 https://github.com/normaldotcom/cantact-fw
[    7.839350] slcan: serial line CAN interface driver
[    7.848965] slcan: 10 dynamic interface channels.
[    7.872546] can0: renamed from slcan0
[    8.039720] F2FS-fs (sda): Can't find valid F2FS filesystem in 1th superblock
[    8.057561] F2FS-fs (sda): Can't find valid F2FS filesystem in 2th superblock
[   21.117029] can0: renamed from slcan0
[   21.377152] can: controller area network core
[   21.404103] can: raw protocol
[   21.870327] can0: renamed from slcan0
root@raspberrypi2:~# udevadm info --query=all --name=/dev/ttyACM0
P: /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2.3/1-1.2.3:1.0/tty/ttyACM0
N: ttyACM0
S: serial-starter/ttyACM0
S: serial/by-id/usb-Protofusion_Labs_CANable_1205aa6_https:__github.com_normaldotcom_cantact-fw_0022000B524E431120373533-if00
S: serial/by-path/platform-3f980000.usb-usb-0:1.2.3:1.0
S: ttycan0
E: DEVLINKS=/dev/serial-starter/ttyACM0 /dev/serial/by-id/usb-Protofusion_Labs_CANable_1205aa6_https:__github.com_normaldotcom_cantact-fw_0022000B524E431120373533-if00 /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2.3:1.0 /dev/ttycan0
E: DEVNAME=/dev/ttyACM0
E: DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2.3/1-1.2.3:1.0/tty/ttyACM0
E: ID_BUS=usb
E: ID_MODEL=CANable_1205aa6_https:__github.com_normaldotcom_cantact-fw
E: ID_MODEL_ENC=CANable\x201205aa6\x20https:\x2f\x2fgithub.com\x2fnormaldotcom\x2fcantact-fw
E: ID_MODEL_ID=60c4
E: ID_PATH=platform-3f980000.usb-usb-0:1.2.3:1.0
E: ID_PATH_TAG=platform-3f980000_usb-usb-0_1_2_3_1_0
E: ID_REVISION=0200
E: ID_SERIAL=Protofusion_Labs_CANable_1205aa6_https:__github.com_normaldotcom_cantact-fw_0022000B524E431120373533
E: ID_SERIAL_SHORT=0022000B524E431120373533
E: ID_TYPE=generic
E: ID_USB_CLASS_FROM_DATABASE=Communications
E: ID_USB_DRIVER=cdc_acm
E: ID_USB_INTERFACES=:020201:0a0000:
E: ID_USB_INTERFACE_NUM=00
[... 53 lines cut ...]
E: ID_BUS=usb
E: ID_MODEL=CANable_1205aa6_https:__github.com_normaldotcom_cantact-fw
E: ID_MODEL_ENC=CANable\x201205aa6\x20https:\x2f\x2fgithub.com\x2fnormaldotcom\x2fcantact-fw
E: ID_MODEL_ID=60c4
E: ID_PATH=platform-3f980000.usb-usb-0:1.2.3:1.0
E: ID_PATH_TAG=platform-3f980000_usb-usb-0_1_2_3_1_0
E: ID_REVISION=0200
E: ID_SERIAL=Protofusion_Labs_CANable_1205aa6_https:__github.com_normaldotcom_cantact-fw_0022000B524E431120373533
E: ID_SERIAL_SHORT=0022000B524E431120373533
E: ID_TYPE=generic
E: ID_USB_CLASS_FROM_DATABASE=Communications
E: ID_USB_DRIVER=cdc_acm
E: ID_USB_INTERFACES=:020201:0a0000:
E: ID_USB_INTERFACE_NUM=00
E: ID_USB_SUBCLASS_FROM_DATABASE=Abstract (modem)
E: ID_VENDOR=Protofusion_Labs
E: ID_VENDOR_ENC=Protofusion\x20Labs
E: ID_VENDOR_ID=ad50
E: MAJOR=166
E: MINOR=0
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=45656549
E: VE_SERVICE=ignore

root@raspberrypi2:~#

kwindrem, 2022-06-20:
I just tried my CANable Pro on my RPI 4 v1.2 board running Venus OS v2.90~12 and it is detected properly at boot and after replugging.

I recall having issues plugging the CANable into a POWERED USB hub not being recognized after boot. This was due to the way the CANable Pro dealt with a USB reset. Without the hub, USB power is removed briefly during boot but with a powered hub it was not.

It's been a while since I used the CANable Pro as I've moved to Cerbo, but I also recall a firmware update that may have fixed the issue with powered hub. dimes indicates I'm running 8c005eb which is a beta version. Check https://canable.io/updater/.

andyalford, 2022-06-21:
Quick answer, thanks! :-)

I tried and confirmed....
1. It was indeed attached to a USB Hub (not powered), so I tried moving it direct to Pi and same problem.
2. I updated the firmware, and it still fails on both the hub connection and direct connection.

Same issue.  It disappears from the Devices List on a restart, even though the blue lights are still flashing.  And if unplugged and reinserted, it appears in the list and all works as expected. 

Shall I send other files?  /u-boot/config.txt or can0something???

kwindrem, 2022-06-21:
The failure is at a low level and you've looked at the dmesg log. There's also /data/log/messages but don't suspect you'd find anything there.

You can play around with the udev rules and related shell scripts to see if you can get a reliable connection on boot. My knowledge there is very limited and I'm afraid I can't offer much help. If you find something that works, let me know and I'll roll it into my install script.

andyalford, 2022-06-23:
Ok, thanks again for the reply.

I obviously did the setup manually last time, so I will go through my notes and check what your Setup scripts have done and compare them with what worked before.

I guess I can hand modify things and see if I can fix it, then let you know the 'before' and 'after.'

Thanks!

P.S.  I will also post on Victron Community Mods forum to see if anyone else can point me in the right direction. :-)

andyalford, 2022-06-23:
Well, I fixed it.  By comparing all the results on my Pi with what I have done manually before.  And rebooting between each time.  Took a while!!

But the thing that made it work was /etc/venus/canbus_ports had a whole bunch of entries, can0 several times, can1, and even a blank line - from all my attempts to install and remove using /data/VeCanSetup/setup, etc.  

When I manually edited /etc/venus/canbus_ports to contain exactly "can0" and a carriage return, it rebooted and all started as it should.

I don't have a second CAN connection right now, but will have in a few weeks.  So I I guess I will be coming back to this issue and trying out how to get 2x CANables to work!

I notice this entry on Venus Mod Community on this topic, but don't quite understand it yet....
https://community.victronenergy.com/questions/111906/two-can-bus-devices-on-rpi-nmea2000-250kbs-and-diy.html

So, all going well, and if anyone else gets the same problem, try this simple edit.  

Loving my SetupHelper system !!  ;-)