kwindrem/VeCanSetup #5: VeCan install help

mar0997, 2022-07-06:
Thank you for all of the great help you have provided in putting together the various packages.  I have installed setup helper and GUI mods with no issues.  When I went to install VeCan I ran into some issues that may just be my lack of understanding.  I first attempted to install into my Cerbo GX via the micro SD card.  It is stored on the Cerbo but then says that I need to install from the command line.  I downloaded putty and was able to log in with that when I quickly realized that my Linux skills are significantly lacking.  Is there a simple explanation of what I should do from there to get it fully installed?

But first I believe that VeCan can help with my issue but want to make sure.  I originally started out by building the system on my boat with Czone equipment.  I have full NMEA 2000 monitoring of my AC and DC systems.  I added a VECan cable to port the data from NMEA 2000 to the Cerbo.  I can see the data in remote console under VE Can Devices.  But I haven’t figured out how to easily port this data over to Dbus.  I have seen some examples but they aren’t quite what I need to do.  Will VeCan help me accomplish this?

Many thanks for your help.

[image]
[image]
[image]

kwindrem, 2022-07-06:
VeCan setup adds VeCan/CANbus ports to a system with less than 2 ports, so for Cerbo, it won't allow you to do anything since it has two bulit-in ports. VeCan setup is really intended for things like the Raspberry PI which has no built-in CANbus ports or for things like CCGX that has only one.

I have no idea what will make a connection to your other control system. Sorry.

PulithJay, 2024-02-01:
hi @kwindrem i installed your canbus from setuphelper and then used putty to try to install. i pressed I then it asked from something i pressed enter and it got installed but when i go to check from rasp its not there. should i enter another line or number.

kwindrem, 2024-02-01:
Installation is a multi-step process from the command line. After selecting 'install' you need to add interfaces (the add function), select the interface type. There may be additional prompts you need to answer as well.

PulithJay, 2024-02-01:
from venus console i went to ethernet and took the IP Address from there. after i pressed I then also it says no CANbus definitions found. i am not very good at this if you could help me would be highly appreciated.

[image]
[image]
[image]

kwindrem, 2024-02-01:
You want to choose add (a), then the interface number you want to end with. VeCanSetup USB interfaces begin with 10.

You will then be asked to select the interface you are adding from the list presented.

BTW, if you only have one CANbus interface and it is a USB GS lib interface, then I suggest you do not use VeCan Setup and let the operating system configure the interface. You only need VeCanSetup if you have more than one USB interface and want to name them, or a hat board for Raspberry PI.

There is also another type of USB interface that uses slcan. These interfaces can be added with VeCanSetup but are not recommended because they may result in unreliable communication.

PulithJay, 2024-02-01:
Yes I have a 2 channel issolated can hat for raspberry pi. So after A I enter 10 then one of those ports will be vecan then for the next one if I want it as canbus how do I program it. Sorry for the trouble. Thanks again. 

[
[image]
](url)

kwindrem, 2024-02-01:
Then you want to enter (h) instead of a port number.

PulithJay, 2024-02-01:
Ok then for the order port I do the same right. And if I need canbus function do I yet put 10 or someother number since there are two ports. I watched a video of offgrid garadge(from Andy) he used an older version of yours where he had to choose if it was waveshare n so but I never got those to select in this version. You been a great help. Thanks again.

kwindrem, 2024-02-02:
You need to configure both hat ports at the same time. If you select the correct device, you should be asked if you want to activate both ports.

PulithJay, 2024-02-05:
I am so sorry now its asking GS USB device type. I am using a  raspberry pi with a Can HAT extention it's a dual CAN HAT. I have attached screen shot. 
[image]

PulithJay, 2024-02-05:
I got it working after alot of thinking. Its bcz I am not a tech guy.
Sorry for the trouble.

Now I got the can bus showing under services but when I connect the daly
bms can wires CAN H, CAN L and ground it shows me state as error-passive.

PulithJay, 2024-02-05:
Disregard my last email I had change the h and l wires. Its working now.
Thanks.

PulithJay, 2024-02-06:
Hi just found a mistake on the guimod thr min and max cell voltages needs
to be swap. Also remaining battery voltage is not showing. I am using daly
bms n the daly is connected using can bus in the raspberry pi. Remaining
time also not showing. Anyway to fix them.

Thanks in advance.

kwindrem, 2024-02-06:
The code for the battery detail page is correct so if there's an issue it is most likely in the driver for your specific BMS. Others using different BMS have not reported an issue.

(This issue should be in GuiMods repo not here in VeCanSetup repo.)

PulithJay, 2024-03-09:
Hi,

Do you think that you can please make a app for raspberry pi fan function.

Waveshare.com/wiki/fan_hat

I tried all but the fan or the display gets on. If this is possible it
would help to keep the raspberry pi cooler. Ohh this is for venous os for
victron.

PulithJay, 2024-03-29:
hi i have somehow made the waveshare fan to work with venus but i dont know how to upload to github and to make it run with setuphelper. if i send you the files can you make it available in setup helper.