kwindrem/VeCanSetup #6: Venus changes

jhofstee, 2022-07-12:
Just to drop you a note, since Venus v2.90~17 there is no longer a limit for 2 CAN-bus interfaces. Services will now be automatically created for new CAN interfaces. You can name them from udev if you like, see https://github.com/victronenergy/meta-victronenergy/commit/e9ba4487c2997f021e7548ddea5e0b16c15bb46c. And slcan no longer needs the bitrate.

There is a script as well in /opt/victronenergy/venus-platform/svectl to install services.

Feel free to close this issue, since it actually isn't an issue, just a note...

kwindrem, 2022-07-13:
Is there documentation for this somewhere? I'd like to update my VeCanSetup package to support more than 2 CAN ports. Is there a practical limit to the number of CANbus ports?

Thanks.

jhofstee, 2022-07-15:
No better, we have code ;), see below.

There is no longer a need to set the bitrate for slcan with -s.
An optional name can be set for a CAN-bus interface.
There is no longer any limit on the number of CAN-interfaces.

https://github.com/victronenergy/linux/commits/venus-5.10.109
https://github.com/victronenergy/meta-victronenergy/commit/e9ba4487c2997f021e7548ddea5e0b16c15bb46c

kwindrem, 2022-07-18:
Do these changes obsolete VeCanSetup? Sorry, I it isn't clear from the links you sent if VeCanSetup will have any value after v2.90~17 and can't look at the code for this release since it's no longer on the development branch and I didn't snag a copy when it was there.

Just want to do the right thing here.

kwindrem, 2022-07-25:
I've got the hot-plug working again in v2.90~18. I added a udevadm trigger line to the add script. Might not be the right way but it's working.

Interface names aren't working. According to comments by jhofstee I should be able to add VE_NAME to the udev rules but this isn't changing the name seen in the Service menu. Still just the port identifier (can0).

I also noticed that udevadm info is no longer showing parameters set in the udev rules file or provided by the device itself. VeCanSetup used to report things like VENDOR_ID but there's no longer anything in what udevadm info provides. I went back to Venus OS 2.73 and the information is still missing back that far.

claib, 2022-08-13:
Hi there
I'm playing around with my 2.9  22 version on my Raspi.
Done the script steps. Seems to be fine. 
[image]
On the Remote Web Interface i see no Service ....
Are there compat issues ?
best
Christoph

kwindrem, 2022-08-13:
The "latest" version of VeCanSetup is not compatible with v2.90 due to the changes in the way Venus OS now handles CANbus ports.

There is a beta version of VeCanSetup available but I'm still working on several issues. I have not had the opportunity to test with any of the CANbus hats but have reports of them working pre v2.90 for others.

claib, 2022-08-14:
thank you very much for your valuable time and effort !