kwindrem/VeCanSetup #7: Venus 2.9 released

claib, 2022-09-17:
Hi :-)

i see 2.9 release is out : yssss
what about the can setup scripts ?
do you advice to do the configuration handish, or will there be the magic sreipt available soon ?
best regards

kwindrem, 2022-09-17:
I am in the process of updating VeCanSetup to support v2.90.

It's been a huge project as many things have changed:

Starting with v2.90, an unlimited number of CANbus ports are permitted.

Venus OS automatically configures GS USB CANbus interfaces without any help from a script. Raspberry PI hats still need manual configuration (overlay instructions in /u-boot/config.txt). canX ports are assigned in the order ports are detected.

Automatic assignments can change from boot to boot.

Hats must be assigned to can0 and can1 but these conflict with automatic GS USB configurations that start with can0.

Addition of udev rules can stabilize the assignments but need to avoid can0 and can1.

USB interfaces that rely on slcan are discouraged as they can drop packets and their speed can't be changed after the interface is brought up.

These will all be addressed in VeCanSetup v3.0. Please be patient.

Until then, here are a few hints to configure manually:

For Raspberry PI hats, use a udev rule like this:
ACTION=="add", SUBSYSTEM=="net",\
  KERNELS=="spi0.0",\
  NAME="can10",\
  ENV{VE_NAME}="anything you want here -- appears in Services menu",\
    "
You need one rule for each hat port on dual-channel boards

For GS USB interfaces:

ACTION=="add", SUBSYSTEM=="net",\
  ENV{ID_VENDOR_ID}=="VENDOR_PH",\
  ENV{ID_MODEL_ID}=="MODEL_PH",\
  ATTRS{serial}=="SERIAL_PH",\
  NAME="can11",\
  ENV{VE_NAME}="anything you want here -- appears in Services menu"

Replace the ..._PH with numbers specific for your interface

slcan USB interfaces are more complicated and beyond the scope if this brief note

Note that interfaces like the CANable USB can be refreshed to use Candelight firmware which is a GS USB protocol and is highly recommended over using slcan.

kwindrem, 2022-09-18:
I just released v3.0~1 (a beta test version) of VeCanSetup that should address the needs for Venus OS v2.90.

I have tested it on v2.90 with CANable Candelight and slcan firmware as well as the Waveshare dual-channel CANbus hat (the non FD version).

This version has code for Venus OS v2.89 and earlier but has not been tested.
I have also not tested other interfaces.

You can download this version using PackageManager by setting the branch to 'beta'.

kwindrem, 2022-09-18:
v3.0~3 is now available for beta test.

Fixed several bugs associated with Venus OS prior to v2.90 where only two CANbus ports were allowed.

Changed the UI for adding / deleting interfaces.

Also confirmed adding a USB CANbus interface on Cerbo v2.90 works.