kwindrem/VeCanSetup #8: PiCAN-M support

PierreSmits, 2022-10-04:
Adjust installation script for the PiCAN-M hat, so that the hat can be selected when using the VeCanSetup setup script, and both the can0 (for the Micro-C/nmea2000 connector) and can1 (for the RS422/nmea0183 connector) are configured

With an installation of the hat on a pi 3b+ (with an openplotter noobs install) the following instructions were followed to get the Pi up and running: 
https://cdn.shopify.com/s/files/1/0563/2029/5107/files/pican-m_UGB_20.pdf?v=1619008196

kwindrem, 2022-10-04:
I'll try but need lots of information:

How did you configure the serial port and get it attached to a can port? Your last comments in the Victron community form indicated you hadn't gotten this working. Maybe I misunderstand that you are trying to make this a second can port. Please provide some additional description of how you are using this

It appears this hat uses the serial port (GPIO 14 and 15) which could conflict with use for the Venus OS console. But this can be enabled/disabled in cmdline.txt if that is what's being used. Let me know because I need to build a GPIO usage for this hat to check for GPIO conflicts.

Did you need to modify /etc/inittab as Mark Bath suggested?

A bit of research indicates the dtoverlay=spi-bcm2835-overlay line is not needed. Can you confirm things work if that is NOT in config.txt?

Does this hat actually use the i2c bus or is it just passing it on for optional use by other devices? Use of the i2c bus conflicts with reassignments I've made in gpio_list to open up pins for SPI interrupts and chip selects. I don't think these will be a conflict with this hat but not 100% sure.

PierreSmits, 2022-10-11:
Hi Kevin,

My apologies for the late reaction. Have been a bit under the weather lately, preventing me to actively work this issue and the one in the Victron community.

> How did you configure the serial port and get it attached to a can port? Your last comments in the Victron community form indicated you hadn't gotten this working. Maybe I misunderstand that you are [...]

I haven't pursued this, as per the suggestion in the Victron thread to first focus on the can issue first. 
The main goal is to get the CANbus through the nmea2000 connector going to get gps and wind data from the network. I can test that here at home as I have a test nmea2000 setup with a gps sensor. 
And after that have the nmea0183 through the rs connector going as I still have depth and speed through water in my boat on that protocol. However, can't test any of that nmea0183 at the moment here at home.

> dtoverlay=spi-bcm2835-overlay
It indeed doesn't seem to be needed. Even on standard pi 3b+ with the setup per supplier's instructions it shows that it is not loaded/used.

PierreSmits, 2022-10-12:
@kwindrem I have made some additional progress on getting the nmea2000 connector working on the pi/venusos setup. 

It appeared to be related to the overlay definition that gets written to the /u-boot/config.txt file.
As it appears in v2.91, the files listed under /u-boot/overlays for the mcp2515 driver don't have the addition -overlay in the filename.
Changing line in the config.txt file from
`dtoverlay=mcp2515-can0-overlay,oscillator=16000000,interrupt=25`
to
`dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25`

and performing a reboot gave me can0 visible in the gui under services