kwindrem/VeCanSetup #9: Waveshare Hat added - but no new interface found

regda1986, 2022-10-05:
Hello,

I have an issue with the setup of an Waveshare 1-channel CANbus Hat. Maybe its just an beginner mistake, but I have no more ideas.

I have added the hat to my raspberry pi 4 2GB and install both packages. 

Than I add the hat in VeCanSetup and the interface is setted up:
CANbus interface definitions:
interface: hat ) Waveshare 1-channel CANbus Hat 12 MHz crystal
  canbus1 (can8) can8 spi0.0

But the problem is, there is no new interface added to the system and i found no can bus port under services in Venus OS:

root@raspberrypi4:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,DYNAMIC,UP,LOWER_UP> mtu 1450 qdisc mq state UP group default qlen 1000
    link/ether e4:5f:01:d5:6f:e3 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.7/24 brd 192.168.178.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::e65f:1ff:fed5:6fe3/64 scope link
       valid_lft forever preferred_lft forever
3: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
4: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,DYNAMIC,UP> mtu 1450 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether e4:5f:01:d5:6f:e4 brd ff:ff:ff:ff:ff:ff
5: ll-eth0@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1450 qdisc noqueue state UP group default qlen 1000
    link/ether 2a:ba:ee:9f:51:b0 brd ff:ff:ff:ff:ff:ff
    inet 169.254.10.40/16 scope link ll-eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::28ba:eeff:fe9f:51b0/64 scope link
       valid_lft forever preferred_lft forever

[image]

Many thanks for your help.

kwindrem, 2022-10-05:
Hat configuration is not always automatic as there are many variables. I have only tested VeCanSetup with a limited number of devices and you 1-channel Waveshare board was not one I tested.

You need to look at the documentation for your hat. Waveshare provides a wiki for each product that will include setup instructions. Make sure VeCanSetup is doing the correct thing. If not, you may need to manually do some of the steps it does like editing the config.txt file to add the proper overlay.

If something is missing, let me know the specifics or at least the model of the hat you are using and I'll try to update VeCanSetup.

Good luck.

regda1986, 2022-10-07:
Thanks for your support!

I have exact these Can hat as 12MHz version: https://www.waveshare.com/wiki/RS485_CAN_HAT

The overlay is correct like in the waveshare wiki. But is it a problem that the overlay is configured as can0 and VeCanSetup create them as an can8 port?

kwindrem, 2022-10-07:
VeCanSetup "moves" an interface from can0 or can 1 to another port to avoid conflicts with automatically configured interfaces which is now part of Venus OS. A USB interface may initialize to can0 before the hat does. If this happens the CANbus hat won't function. The reverse could also happen if a USB interface is plugged into the system after boot.

So hat CANbus ports are moved to can8 and can9 and USB interfaces are moved to ports starting at can10.

If you need to retain the interface on can0 then you can not use VeCanSetup and will need to build your own udev rules.