kwindrem/VeCanSetup #10: not an real issue; need an case to 3dprint for raspi with canhat

gerdluthe, 2022-10-07:
does anyone got an file to print a case for an raspberry pi with waveshare canhat onboard ?

sorry little bit offtopic but didn't know where to post else

regards

kwindrem, 2022-10-08:
VeCanSetup has two templates for Waveshare single-channel CANbus hats (8 and 12 MHz).
If you have a specific model, I can look into it or you can make your own template.

gerdluthe, 2022-10-08:
[[]](https://www.waveshare.com/rs485-can-hat.htm)(url)

i search a case for this one

gerdluthe, 2022-10-08:
https://www.waveshare.com/rs485-can-hat.htm

kwindrem, 2022-10-08:
Use Waveshare 1-channel CANbus Hat 12 MHz crystal for this one

gerdluthe, 2022-10-08:
No, that's an misunderstanding. i need a printfile to print a case for the raspberry PI with an can hat installed onto it
:-)

closed 2023-09-17 by gerdluthe