kwindrem/VeCanSetup #11: RS485 and CAN Interface on Wafeshare Hat ?

claib, 2022-10-10:
hi there
can i use both interfaces ?
- and how ?
- i need RS485 and normally using https://github.com/Louisvdw/dbus-serialbattery/ to attach my BMS with a USB adapter.
the Louis script offers the drivers for my JKBMS
any idea ?

kwindrem, 2022-10-10:
VeCanSetup does not address the R485 configuration so you'd need to do this manually. Another user had similar questions regarding the PiCan-M interface. Not sure what he eventually ended up with:

Looking at the schematic, the RS485 portion uses the PI's uart pins and should not conflict with the CANbus portion of the hat. This may conflict with the Venus OS use of the uart for it's console so this may need to be disabled. A note in the wiki for this product also mentions the RPI uses it's uart for bluetooth which could also create conflicts in Venus OS.

I'm not sure how the RS-485 connections are made. Is it possible they use the same connector as CANbus? If so they could not be used at the same time.

Sadly, the wiki doesn't contain much information regarding the RS-485 part of this module.

https://github.com/kwindrem/VeCanSetup/issues/8

There is also a discussion here that may be a help:

https://community.victronenergy.com/questions/159239/pican-m-pi-3b-venusos.html

If this ever gets resolved so that the RS485 portion works, I can probably add the necessary configuration and templates to VeCanSetup but can't do this without the help of others that actually have one of these boards

herrfrei, 2023-08-13:
If you are talking about [this device](https://www.waveshare.com/rs485-can-hat-b.htm), you might check out my new package: https://github.com/herrfrei/VeWsCan485