kwindrem/VeCanSetup #13: Not a real issue just a note on Canbus

RobDuthie, 2022-10-12:
Re the latest Venus version 2.92-1 you dont have to install the Vecanbus mod as it starts automaticly when you plug in a 
USB canbus interface using the gsusb canable firmware interface.
I have tried this on raspi3  and raspi4 with no issues, it just work out the box. on 4x raspi tested units.

kwindrem, 2022-10-12:
Works fine as long as you don’t mix gs usb and hats. If you mix them both try to grab can0 and only the first works. 

I’m not sure slcan interfaces work without VeCanSetup

VeCanSetup allows you to name interfaces, resolves gs usb / hat, and allows slcan interfaces to be configured for 250 and 500 kb

kwindrem, 2022-10-12:
The other thing that VaCanSetup does is insure multiple interfaces are locked to the same can port. Without it can port assignments depends on discovery order which can change