kwindrem/VeCanSetup #15: V2.90  can - bus install

MS123uk, 2022-12-24:
hi had time too read up re can bus install  first time doing this and U tube info is out date and not for  rasp I 4  and Venus v2.91 

Since Venus v2.90 CAN-bus interfaces will be automatically recognized and services are created for them. There is no longer a need to set them up manually  ?  so no setup  Venus services page i.e. board rate  as I'm can not see  can -bus on the this page ! having installed Waveshare 2-CH can hat none FD  but can not see it in Venus  so trying to work if install is ok 

Venus Venuslogin as: root
root@192.168.0.25's password:
Last login: Fri Dec 23 23:27:08 2022
root@raspberrypi4:~# /data/VeCanSetup/setup
--- starting setup script v3.3

VeCanSetup adds CANbus ports to the Venus OS system
Raspberry PI includes 0 CANbus ports
an unlimited number of additional ports may be added
one single or dual CANbus hat can be configured by this script

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: i

CANbus interface definitions:
interface: hat ) Waveshare 2-channel CANbus (non FD) Hat
  channel 1: can0 (can8) can8 spi0.0
  channel 2: can0 (can8) can9 spi0.1

Available actions:
  Add a USB interface (a <interface>)

this is from setup

kwindrem, 2022-12-24:
The automatic CANbus configuration included in v2.90 only works on GsUSB CANbus interfaces, not slcand or Raspberry PI hats.

Looks OK as far as you took it. Once you get to this point, you enter a blank line and installation will continue. There will be additional questions then eventually you will be asked to reboot the system. After the reboot, you should see the CANbus hat in the Services menus as can8 and can9.

The ReadMe file explains why those interface ports are selected for CANbus hats.

MS123uk, 2022-12-24:
A blank line stopped it from work  thank for exhalent work have a happy Christmas

closed 2022-12-24 by MS123uk