kwindrem/VeCanSetup #16: no CAN-bus under Services; can0 DOWN !

gerd-git, 2022-12-25:
Hi,
I don't see "CAN-bus" under "Services". 

I can see it here, but DOWN:
```
root@raspberrypi2:~# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,DYNAMIC,UP,LOWER_UP> mtu 1450 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:e4:e8:96 brd ff:ff:ff:ff:ff:ff
3: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can
4: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,DYNAMIC,UP> mtu 1450 qdisc pfifo_fast state DOWN mode DORMANT group default qlen 1000
    link/ether b8:27:eb:b1:bd:c3 brd ff:ff:ff:ff:ff:ff
5: ll-eth0@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1450 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether fa:98:86:9f:5b:1d brd ff:ff:ff:ff:ff:ff
```	
"up" is not working
```
root@raspberrypi2:~# ip link set dev can0 up
RTNETLINK answers: Invalid argument

```

What am I doing wrong?

My System:
VeCanSetup v3.3 via Package manager
Waveshare 1-channel CANbus Has 12 MHz crystal
Raspberrypi 2
firmware v2.89

Thanks for answers in advance
Greetings Gerd

kwindrem, 2022-12-25:
see what is in dimsg:

dmesg | grep spi0
dmesg | grep can0
dmesg | grep -i waveshare

see if there are any services running for can0

ps | grep can0

gerd-git, 2022-12-25:
```
root@raspberrypi2:~# dmesg | grep spi0
[    3.980333] mcp251x spi0.0 can0: MCP2515 successfully initialized.
[  138.070031] mcp251x spi0.0 can0: bit-timing not yet defined
[  138.075754] mcp251x spi0.0: unable to set initial baudrate!
[  215.302927] mcp251x spi0.0 can0: bit-timing not yet defined
[  215.308518] mcp251x spi0.0: unable to set initial baudrate!
root@raspberrypi2:~#

root@raspberrypi2:~# dmesg | grep can0
[    3.980333] mcp251x spi0.0 can0: MCP2515 successfully initialized.
[  138.070031] mcp251x spi0.0 can0: bit-timing not yet defined
[  215.302927] mcp251x spi0.0 can0: bit-timing not yet defined
root@raspberrypi2:~#

root@raspberrypi2:~# dmesg | grep -i waveshare
root@raspberrypi2:~#

root@raspberrypi2:~# ps | grep can0
  978 root      1612 S    supervise can-bus-bms.can0
  984 root      1612 S    supervise dbus-motordrive.can0
  991 root      1612 S    supervise dbus-valence.can0
 9093 root      2692 S    grep can0
root@raspberrypi2:~#

```

kwindrem, 2022-12-25:
Did you set up can0 using VeCanSetup? 

Look in /etc/udev/rules.d for VeCanSetup.rules If that is not present, VeCanSetup didn't install the necessary stuff to initialize the interface.

You must run /data/VeCanSetup/setup from the command line. Simply attempting to install it from PackageManager won't work unless you have run it first from the command line and setup the interfaces.

I see the can-bus-bms service running for can0 but not the main vecan-dbus

VeCanSetup's setup script will also log it's activity in /data/log/SetupHelper. You can look there to see if there are any problems.

gerd-git, 2022-12-25:
I had run VeCanSetup from the command line after installation via PackageManager.

```
root@raspberrypi2:~# ls -la /etc/udev/rules.d
drwxr-xr-x    2 root     root          1024 Dec 25 14:27 .
drwxr-xr-x    6 root     root          1024 Jul 15 12:36 ..
-rw-r--r--    1 root     root             0 Jul 15 12:06 80-net-name-slot.rules
-rw-r--r--    1 root     root           394 Dec 25 14:27 VeCanSetup.rules
-rw-r--r--    1 root     root             0 Dec 25 14:27 VeCanSetup.rules.NO_ORIG
-rw-r--r--    1 root     root           847 Jul 15 12:28 automount.rules
-rw-r--r--    1 root     root           757 Jul 15 12:28 autonet.rules
-rw-r--r--    1 root     root           165 Jul 15 12:28 bluetooth.rules
-rw-r--r--    1 root     root           159 Jul 15 12:28 ether.rules
-rw-r--r--    1 root     root           885 Jul 15 12:06 local.rules
-rw-r--r--    1 root     root           843 Jul 15 12:28 localextra.rules
-rw-r--r--    1 root     root           644 Jul 15 12:28 machine.rules
-rw-r--r--    1 root     root            63 Jul 15 12:28 mtd.rules
-rw-r--r--    1 root     root            51 Jul 15 12:28 rfkill.rules
-rw-r--r--    1 root     root          1634 Jul 15 11:58 serial-starter.rules
-rw-r--r--    1 root     root           590 Jul 15 12:28 simcom.rules
-rw-r--r--    1 root     root           319 Jul 15 12:23 tanksensor.rules
root@raspberrypi2:~# cat /etc/udev/rules.d/VeCanSetup.rules
# udev rules for Waveshare 1-channel CANbus Hat

# replace ...PH (place holder) with actual values
# DELETE the placeholder lines if they aren't used !!
# rename file to e.g., can0.rules

# find interface and connect directly to CANbus port
ACTION=="add", SUBSYSTEM=="net",\
  KERNELS=="spi0.0",\
  NAME="can0",\
  RUN+="/usr/bin/logger [udev] CANbus device added:\
    can0\
    spi0.0\
    "
root@raspberrypi2:~#
```

gerd-git, 2022-12-25:
```
root@raspberrypi2:~# cat /data/log/SetupHelper
@40000000631b985d236a372c blind install rcS.local: start
@40000000631b985d2a04b8b4 blind install rcS.local: running blindInstall.sh
@40000000631b985d2a6a2d5c blind install rcS.local: completed
@40000000631b985d2ebb7e4c blindInstall/blindInstall.sh: starting
@40000000631b985f03630314 blindInstall/blindInstall.sh: waiting for dBus settings
@40000000631b9861035e2114 blindInstall/blindInstall.sh: waiting for dBus settings
@40000000631b986f21ccabfc blindInstall/blindInstall.sh: waiting for dBus settings
@40000000631b98731d20719c blindInstall/blindInstall.sh: moving SetupHelper archive into position
@40000000631b98731e59860c blindInstall/blindInstall.sh: removing rcS.local from blind install
@40000000631b98731f1c7c34 blindInstall/blindInstall.sh: installing SetupHelper
@40000000631b987333944ad4 SetupHelper/setup: --- starting setup script v4.23
@40000000631b9874013a75f4 SetupHelper/setup: creating root setup options directory /data/setupOptions
@40000000631b98740215f944 SetupHelper/setup: creating package options directory /data/setupOptions/SetupHelper
@40000000631b98740315ae34 SetupHelper/setup: creating file set for v2.82-large-30
@40000000631b987421d9c774 SetupHelper/setup: installing PackageManager service
@40000000631b987e26617a44 SetupHelper/setup: adding SetupHelper/setup to reinstallScriptsList
@40000000631b987e26db49dc SetupHelper/setup: creating /data/rcS.local
@40000000631b987e283e0a44 SetupHelper/setup: completed - restarting GUI
@40000000631b987e2939c3ac blindInstall/blindInstall.sh: completed
@40000000631b99743296a154 RpiDisplaySetup/setup: --- starting setup script v2.8
@40000000631b9974390db48c RpiDisplaySetup/setup: creating package options directory /data/setupOptions/RpiDisplaySetup
@40000000631b997439c6d714 RpiDisplaySetup/setup: creating file set for v2.82-large-30
@40000000631b99750430edec RpiDisplaySetup/setup: installing RPI Display Backlight overlay
@40000000631b997504f1ffb4 RpiDisplaySetup/setup: install options have not been set - installng overlay and rebooting for auto discovery
@40000000631b9975064f12fc RpiDisplaySetup/setup: adding RpiDisplaySetup/setup to reinstallScriptsList
@40000000631b997507d5d23c RpiDisplaySetup/setup: script will run again at startup
@40000000631b998427e95e7c SetupHelper/reinstallMods: reinstallMods starting
@40000000631b9986003152f4 SetupHelper/reinstallMods: waiting for dBus settings
@40000000631b99883a5d6224 SetupHelper/reinstallMods: waiting for dBus settings
@40000000631b9999342a87c4 RpiDisplaySetup/setup: --- starting setup script v2.8
@40000000631b999a03d963c4 RpiDisplaySetup/setup: auto-discover display type RPI
@40000000631b999a046490ac RpiDisplaySetup/setup: ++ Installing display enhancements for display: RPI
@40000000631b999a06c7021c RpiDisplaySetup/setup: adding disable_overscan
@40000000631b999a07aabee4 RpiDisplaySetup/setup: adding hdmi_blanking
@40000000631b999a0a0c9414 RpiDisplaySetup/setup: updating config.txt
@40000000631b999a0a798a9c RpiDisplaySetup/setup: setting display blanking to /sys/class/backlight/rpi_backlight/bl_power
@40000000631b999a0af4293c RpiDisplaySetup/setup: setting up brightness control: /sys/class/backlight/rpi_backlight
@40000000631b999a0d778c44 RpiDisplaySetup/setup: disabling AutoBrightness
@40000000631b999b25abf584 RpiDisplaySetup/setup: resetting /Settings/Gui/Brightness max for the RPI display (255)
@40000000631b999c20daaadc RpiDisplaySetup/setup: activating local GUI
@40000000631b999c248207fc SetupHelper/reinstallMods: RpiDisplaySetup reinstall requested reboot
@40000000631b999c24f74b84 SetupHelper/reinstallMods: rebooting ...
@40000000631b999c36861394 SetupHelper/reinstallMods: reinstallMods complete
@40000000631b99a327f197f4 SetupHelper/reinstallMods: reinstallMods starting
@40000000631b99a32d00be8c SetupHelper/reinstallMods: reinstallMods complete
@40000000631ba3062834bb74 SetupHelper/reinstallMods: reinstallMods starting
@40000000631ba3062d69f80c SetupHelper/reinstallMods: reinstallMods complete
@40000000631ba3062727b45c SetupHelper/reinstallMods: reinstallMods starting
@40000000631ba3062c5a0eac SetupHelper/reinstallMods: reinstallMods complete
@40000000631ba30627d78fe4 SetupHelper/reinstallMods: reinstallMods starting
@40000000631ba3062ceb428c SetupHelper/reinstallMods: reinstallMods complete
@40000000631c59bf283dc7dc SetupHelper/reinstallMods: reinstallMods starting
@40000000631c59bf2d6646bc SetupHelper/reinstallMods: reinstallMods complete
@40000000631c59bf293022d4 SetupHelper/reinstallMods: reinstallMods starting
@40000000631c59bf2e440074 SetupHelper/reinstallMods: reinstallMods complete
@40000000631c59bf282e00cc SetupHelper/reinstallMods: reinstallMods starting
@40000000631c59bf2d52a364 SetupHelper/reinstallMods: reinstallMods complete
@40000000631c59bf27f1f5b4 SetupHelper/reinstallMods: reinstallMods starting
@40000000631c59bf2d09de7c SetupHelper/reinstallMods: reinstallMods complete
@40000000631c59bf27f489dc SetupHelper/reinstallMods: reinstallMods starting
@40000000631c59bf2d189bec SetupHelper/reinstallMods: reinstallMods complete
@40000000631c59bf27a9cd0c SetupHelper/reinstallMods: reinstallMods starting
@40000000631c59bf2ce1a59c SetupHelper/reinstallMods: reinstallMods complete
@40000000631c59bf280cfbac SetupHelper/reinstallMods: reinstallMods starting
@40000000631c59bf2d35dc34 SetupHelper/reinstallMods: reinstallMods complete
@40000000631c59bf27954b0c SetupHelper/reinstallMods: reinstallMods starting
[... cut ...]
@4000000063a85dea231d6b04 VeCanSetup/setup: removing scripts from old version of VeCanSetup
@4000000063a85dea27133e14 VeCanSetup/setup: adding udev rules for hat
@4000000063a85dea27f14204 VeCanSetup/setup: updating VeCanSetup udev rules
@4000000063a85dea29c52744 VeCanSetup/setup: updating canbus_ports file
@4000000063a85dea2c4628ec VeCanSetup/setup: adding new CAN hat overlay to config.txt
@4000000063a85e4e29a5859c SetupHelper/reinstallMods: reinstallMods starting
@4000000063a85e4e33266dd4 SetupHelper/reinstallMods: reinstallMods complete
root@raspberrypi2:~#

```

kwindrem, 2022-12-25:
Doesn't look like my script is installing the correct services: creating vecan-dbus.hat.config. should be creating vecan-dbus.can0. I'll work on a fix

gerd-git, 2022-12-25:
Thank you! I am waiting for your fix.

kwindrem, 2022-12-25:
I think I fixed it. Made several changes that involve the way the interface is defined and stored in the config directory so I would delete any existing interfaces then add them back in.

Let me know how it goes.

gerd-git, 2022-12-26:
after update to v3.4 via package Manager, still no CAN-bus under Services:

```
root@raspberrypi2:~# cat /u-boot/config.txt
kernel=u-boot.bin

[pi2]
device_tree=bcm2709-rpi-2-b.dtb

[pi3]
device_tree=bcm2710-rpi-3-b.dtb
dtoverlay=miniuart-bt
core_freq=400
core_freq_min=400

[pi3+]
device_tree=bcm2710-rpi-3-b-plus.dtb

[all]
dtparam=spi=on

# no overscan
disable_overscan=1

# allow HDMI display to enter low powr mode
hdmi_blanking=1
#### RpiDisplaySetup begin
disable_overscan=1
lcd_rotate=2
#### RpiDisplaySetup end
#### begin CAN overlay
# Waveshare 1-channel CANbus Hat 12 MHz crystal
# instantiates as can0 ONLY !!!
# use udev rules to rename
[all]
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=12000000,interrupt=25,spimaxfrequency=2000000
#### end CAN overlay
root@raspberrypi2:~#
root@raspberrypi2:~# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,DYNAMIC,UP,LOWER_UP> mtu 1450 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:e4:e8:96 brd ff:ff:ff:ff:ff:ff
3: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can
4: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,DYNAMIC,UP> mtu 1450 qdisc pfifo_fast state DOWN mode DORMANT group default qlen 1000
    link/ether b8:27:eb:b1:bd:c3 brd ff:ff:ff:ff:ff:ff
5: ll-eth0@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1450 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether a2:2a:a8:29:c0:99 brd ff:ff:ff:ff:ff:ff
root@raspberrypi2:~#
root@raspberrypi2:~# /data/VeCanSetup/setup
--- starting setup script v3.4

VeCanSetup adds CANbus ports to the Venus OS system
Raspberry PI includes 0 CANbus ports
2 additional ports may be configured
one single CANbus hat may be configured

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: u

CANbus interface definitions:
interface: hat ) Waveshare 1-channel CANbus Hat 12 MHz crystal
[... 130 lines cut ...]
  Add a USB interface (a <interface>)
  Delete existing interface (d <interface>)
interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 0 or 1
Hat interface should be entered as 'h' or 'hat'

Choose action from list above or (cr) if done with changes:
installing CANbus ports
removing scripts from old version of VeCanSetup
adding udev rules for hat
updating VeCanSetup udev rules
updating canbus_ports file
adding new CAN hat overlay to config.txt
creating can-bus-bms.can0
creating vecan-dbus.can0
creating mqtt-n2k.can0
creating dbus-valence.can0
creating dbus-motordrive.can0

new interfaces will not become active until after a reboot
Reboot system now (y) or do it manually later (n): y
```
Now there ist a "CAN-bus (1)" under services!

THANK YOU!!!!!!