kwindrem/VeCanSetup #17: PICAN-M installation

dahlei, 2023-01-25:
I am trying to add the PiCAN M to Venus OS large

See snipp from the setup: /data/VeCanSetup/setup
Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: i

no CANbus definitions found
Available actions:
  Add a USB interface (a <interface>)
interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above or (cr) if done with changes:

I am using RPi 4 with the Venus OS large. I have the SetupHelper, ShutdownMonitor, VeCanSetup, RpiGpioSetup and GuiMods.
I am using GPIO 17, 22, 23, 24 and 27 for controlling relays, this is working perfectly

kwindrem, 2023-01-25:
Not sure what your problem is. From the snippet, you haven't gone far enough in the installation procedure to select the hat board.

I looked and there is a PiCan-M template but it's untested.

Also, the RS-485 interface is NOT part of this setup. You'll need to do this manually.

dahlei, 2023-01-25:
What is the installation procedure for the PiCAN-M template? To this point I've been abel to follow the instructrions, now I am struggeling. There is no blind install or script to run for this?

kwindrem, 2023-01-25:
You must install from the console as there are too many choices. The original post indicates you are doing this but didn't take it far enough. If you follow the prompts, it will guide you through the process.

Where you stopped, it was asking to add an interface. You should answer "a hat".
You will then be presented with a choice of interfaces. Choose the PiCan hat by number
There will be additional questions that you need to answer and eventually get back to the add interface prompt. Since you have nothing else to add enter nothing and the interface will be configured and you will be prompted to restart.

dahlei, 2023-01-25:
Thank you! Got it going now!