kwindrem/VeCanSetup #18: Regular "BMS connection lost error" (#67) with two MPTTs

joesolar, 2023-03-21:
Hi, 

I am using VeCanSetup successfully with my Venus on Raspi with a REC BMS 16 (Victron) and 2 USB MPPT Trackers and a standard CAN HAT.

In general , eyerything runs smoothly BUT I am regularly an error #67 (BMS connection lost) from one of the two connected MPPTs.

- There is no general rule, which one is connected (one of the two gets the error, the other one remains working properly)
- I cannot reproduce the error. Sometimes it runs a day w/o any problem, sometimes I see the error all 20 minutes
- Re-Connecting the USB device, resetting the MQTT or even rebooting Venus resolves the issue until next time (MQTT switches to standyby after BMS connection is lost and needs manual reset.
- Error was existing with a completely different CAN haat hardware as well, so I am guessing it's a "timeout" or issue with two identical devices connected via USB.

Using CAN8 port, all devices on latest firmware.

Error was existing before FW upgrade, was existing with Venus v2.89 (before the change victron made) but is still existing with current Venus v3.00~21-

[image]

Do you have any clue what the issue with the two identical USB devices yould be? 

Here some MQTT data:

[image]

Help is much appreciated as I cannot reset my devices manually every day in order to get PV production ;)

joesolar, 2023-03-21:
> I have no idea what's going on, but most likely nothing that VeCanSetup is
> causing.
>
> You might check the logs for the services related to the BMS and MPPTs and
> the CANbus BMS service
>
> You might post this issue to the Victron community forum (modifications
> space) also.

Hi,

I did already getting the response "unsupported as using Raspberry"
L.

There are no real logs and the ones I found did not show anything. Also in
contact with REC BMS, mo solution there.

I recognize that you wrote that two identical devices need to be enriched
with SN in order to identify them correctly. Do the included scripts add
the necessary data automatically?

kwindrem, 2023-03-21:
Serial numbers aid in uniquely identifying USB-based CAN-bus interfaces. Since you are using a single channel CANbus interface you won't need a serial number for uniqueness. Also, you are using a hat so there's probably no serial number anyway. I believe my script only asks for and suggests serial numbers for USB interfaces.

joesolar, 2023-03-22:
Good morning!

The REC BMS is connected via the CAN HAT, but the two MPPT-Trackers are connected via USB to the Raspi. So the "connection" between the BMS and the MPPTs is handled in the Raspberry device via VENUS OS.

I could found this in the CAN log:

@4000000063f9529d181a305c can-bus-bms v0.39
@4000000063f9b53c3571a5f4 *** CCGX booted (0) ***
@4000000063f9b5541e76d20c *** starting can-bus-bms ***
@4000000063f9b554307c123c can-bus-bms v0.39
@4000000063f9c2111e1433f4
@4000000063f9c2111e14477c -------- dumping backlog -------
@4000000063f9c2111e145334 INF.canhw_drv: adding driver socketcan
@4000000063f9c2111e145eec INF.: Using can8
@4000000063f9c2111e1462d4 INF.can-bms: Product ID from settings: B022
@4000000063f9c2111e147274 INF.can-bms: dbus_service: registering name com.victronenergy.battery.socketcan_can8
@4000000063f9c2111e148214 INF.can-bms: Manufacturer: 'REC-BMS'
@4000000063f9c2111e148dcc **ERR.bms: bms timeout**
@4000000063f9c2111e14959c INF.can-bms: disconnecting from dbus
@4000000063f9c21c1d87518c INF.can-bms: dbus_service: registering name com.victronenergy.battery.socketcan_can8
@4000000063f9c21c2764dc9c INF.can-bms: Manufacturer: 'REC-BMS'

Probably you tell me again that this is not an issue of VeCanSetup - but appreciate your thougths if you have any, where the problem yould sit as it seems your're an expert and I am not ;)

Thanks in advance!

kwindrem, 2023-03-22:
This is something REC and Victron will need to work out. Either the timeout in the GX device needs to be lengthened or REC needs to increase the frequency of it's reporting.

You should post this issue to the Victron community forum to get Victron's attention. It also may catch the attention of others using this BMS who could then provide additional help.

With the single CANbus interface in your system, you can uninstall VeCanSetup then install the line in /u-boot/config.txt to activate the interface's DT overlay. This will put the interface on can0 but I do not expect the behavior to change.

You might try switching to a different CANbus interface, either a different hat or batter yet a USB interface that uses Gs USB protocol (not slcan). I don't know if this will solve the issue but for about $30 it may be worth the experiment.