kwindrem/VeCanSetup #21: VenusOS Beta V3.00~36, VeCanSetup doesn‘t find any CAN bus definition

raspi4all, 2023-04-25:
I changed my ESS from rpi4 VenusOS V2.89 to V3.00~36. I run VeCanSetup to re-add Pylontechs BMS connected to Waveshare Dual CAN HAT, but VeCanSetup didn't find any CAN bus definitions. In V2.89 of VenusOS it was straight forward with VeCanSetup. I'm wondering, that actually under VenusOS V3.00 the ESS is working without any CAN bus setup for pylontech BMS. Did VenusOS V3.00~36 automatically integrate Pylontech BMS without any CAN HAT setup? I don't think so.

kwindrem, 2023-04-26:
You would still need a CANbus interface. Venus OS handles USB interfaces without VeCanSetup but not hats.

I checked VeCanSetup on v3.00~36 on RPI 4 and it worked fine.

raspi4all, 2023-04-26:
It‘s totally clear that I need a CAN Bus interface. The whole system did already run for a while with VenusOS V2.89 and a Waveshare Dual CAN HAT interfacing the Pylontech BMS. Now, with the Beta I‘ve installed SetupHelper and VeCanSetup as I already did for the V2.89 system, but VeCanSetup said that it couldn‘t find any CAN Bus definitions. The cmdline menu is empty, where I normally would have a choice what kind of interface I‘d like to install.

raspi4all, 2023-04-26:
Finally I could edit /u-boot/config.txt by hand as suggested by Waveshare to add the MCP2515 based CAN buses. A second solution would be an USB2CAN Interface. Actually I can‘t imagine any reason that leads VeCanSetup to fail only on my rpi4 V3.00~36. Maybe there is an issue with my SPI driver or sth. else. Is there a VenusOS variant of raspi-config? I can‘t remember, whether I enabled SPI separatly in V2.89.

kwindrem, 2023-04-26:
No raspi-config

VeCanSetup doesn’t look for existing interfaces. Rather it builds the list of interfaces to choose from from definitions in the package directory. If you have not done so re download and reinstall the package

raspi4all, 2023-04-26:
What package did you mean to install before first VeCanSetup run? In my opinion everything  needed to interface the Waveshare CAN HATs is already built-in the VenusOS RPi images. But if VeCanSetup shows the different CAN interface choices at your system, did you install any package except SetupHelper?

closed 2023-04-27 by raspi4all