kwindrem/VeCanSetup #22: Possiblilty to change can bus bitrate or new can bus profile

chaerle, 2023-06-02:
Is there a possibility to change the can bus bitrate to 125.000 kbs for a certain bus? I want to integrate a canbus-charger working at this speed. 

There is already a possibilty in venusos the select a can-bus-profile, but all have 250kbps or 500kps. 

Maybe there is some code needed to add a new can-bus-profile?

Autospark86, 2023-12-23:
i'm wondering the same question.
looking at making a CI-BUS profile

kwindrem, 2023-12-23:
This is beyond what VeCanSetup can do and also beyond my capabilities. It needs to be rated in the Victron community forum (modifications section). So those more in the know of the ins and outs of CANbus in the Venus OS environment can comment on.

Sorry I can't be of more help here.