kwindrem/VeCanSetup #24: Victron MK3 USB to VECan Adapter compatible?

Magnus8513, 2023-07-28:
Hi,

is it possible to use the Victron MK3 USB to VECan adapter to connect to a BMS via can? 

If so, what would I need to do to achieve this. 

Regards
Magnus

kwindrem, 2023-07-28:
I'm not sure. I did not find anything on the Victron product page or in their on-line search about such a product.
A link would help.

Magnus8513, 2023-07-28:
https://www.victronenergy.de/accessories/interface-mk3-usb

kwindrem, 2023-07-28:
Got it. No, this is to for VE.Bus, not VE.Can (aka CANbus).
Victron does sell a USB to VE.Can adapter but if I recall, it's pretty expensive. There are many USB to CANbus adapters on the market for less money. Pick one that supports GS Lib, not SLCAN.