kwindrem/VeCanSetup #25: Venus OS 3.01 not support?

mscodeNEO, 2023-09-02:
Hello,

I cannot add the Waveshare CAN hat. "Install and ACtivate (i)" returns: No can bus definitions found. I cannot add anything.

Am I doing something wrong or do I have to use an older version like Venus OS 2.9?

Thanks in advance
Matthias

Last login: Sat Sep  2 18:35:41 2023
root@raspberrypi2:~# /data/VeCanSetup/setup
--- starting setup script v3.4

VeCanSetup adds CANbus ports to the Venus OS system
Raspberry PI includes 0 CANbus ports
an unlimited number of USB ports may be configured
one single or dual CANbus hat may be configured

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: i

no CANbus definitions found
Available actions:
  Add a USB interface (a <interface>)
interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above or (cr) if done with changes:

kwindrem, 2023-09-02:
There are no Venus OS version dependencies for VeCanSetup. I just tried it on Venus OS v3.01 and it works fine. I'm running VeCanSetup v3.4 also.

Note, you need to specify 'hat' when prompted. If you simply hit cr noting will be added:

--- starting setup script v3.4

VeCanSetup adds CANbus ports to the Venus OS system
Raspberry PI includes 0 CANbus ports
an unlimited number of USB ports may be configured
one single or dual CANbus hat may be configured

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: i

no CANbus definitions found
Available actions:
  Add a USB interface (a <interface>)
interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above or (cr) if done with changes: a
Enter interface: h
1 ) Canberry Hat (1-channel)
2 ) PiCAN-M Rev B 2.0 Hat (1 CAN-bus, 1 RS422 channel)
3 ) PiCAN-M Rev B 2.0 Hat (1 CAN-bus, 1 RS422 channel)
4 ) PiCAN Hat (1-channel)
5 ) Waveshare 1-channel CANbus Hat 12 MHz crystal
6 ) Waveshare 1-channel CANbus Hat 8 MHz crystal
7 ) Waveshare 2-channel CANbus (non FD) Hat
8 ) Waveshare 2-channel CANbus FD Hat

Choose the Hat configuration from the list above (by number) (cr to skip): 7
selected interface: Waveshare 2-channel CANbus (non FD) Hat

Install both channels of this interface (y/n)?: y

Enter optional name for first port on this Hat (cr for none): Can One
Enter optional name for second port on this Hat (cr for none): Can Two
Waveshare 2-channel CANbus (non FD) Hat
  Can One (can8)  spi0.0
  Can Two (can9)  spi0.1

Install interface as shown above (y/n)?: y
adding CANbus hat
Hat configuration updated

CANbus interface definitions:
interface: ( hat ) Waveshare 2-channel CANbus (non FD) Hat
  channel 1: Can One (can8) can8 spi0.0
  channel 2: Can One (can8) can9 spi0.1

Available actions:
  Add a USB interface (a <interface>)
  Delete existing interface (d <interface>)
interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above or (cr) if done with changes: 
installing CANbus ports
removing scripts from old version of VeCanSetup
adding udev rules for hat
updating VeCanSetup udev rules
adding new CAN hat overlay to config.txt

new interfaces will not become active until after a reboot
Reboot system now (y) or do it manually later (n): y
rebooting ...

closed 2023-09-02 by kwindrem

mscodeNEO, 2023-09-03:
I am so sorry. It works fine. I was really confused about choosing USB (but this was the right option). I did not try this first because I did not expected, that the "hat" device is an "USB" device, is it?.

kwindrem, 2023-09-03:
Hats plug into the 26-pin connector on a Raspberry PI. All other interfaces use the USB connection. Victron recommends against USB adapters that use slcan due to poor performance. USB adapters using GS lib are much better and can be auto-discovered by Venus OS without VeCanSetup -- with some limitations: no interface names and interface to port assignments are dependent on discovery order.