kwindrem/VeCanSetup #26: Misslabeled installation instruction with can hat

ansgar-santic, 2023-10-07:
I might have found a instruction spelling mistake. When first installing a new CAN Hat, you need to add the CAN hat and define which one it is. After this the instruction says continue with (cr) but that is listed as invalid action. 
I found out that its supposed to be enter

Shitty Explanation, have a look at my attached Screenshot

[image]

kwindrem, 2023-10-07:
I assumed most people would understand "(cr)" to be just return/enter with no text at the prompt. I'll think of a better way to express this.

ansgar-santic, 2023-10-07:
dayum, well then im the representation of probably the dumb user :D Thanks anyways!

kwindrem, 2023-10-07:
I've updated prompting to hopefully make things more clear:

Available actions:
  Add a USB interface (a <interface>)
  Delete existing interface (d <interface>)
  blank line if done with changes

closed 2023-10-07 by kwindrem