kwindrem/VeCanSetup #28: CAN device not showing

bluey1, 2023-10-11:
Hi
I have installed the VECANSetup, and the CAN device is not showing in the Venus software device page.
The CAN adapter appears to be working.
I am also not getting any error messages on install.
Is there some commands I can use to check the installation
Any help greatly appreciated'
Thank you
Ian T
[image]
[image]

kwindrem, 2023-10-11:
CANbus ports do not show up in the device list.

VeCanSetup shifts all can devices to can8 and above so seeing can0 means the device is not being addressed by the udev rules put in place by VeCanSetup. In VeCanSetup, you must identify the device then set up a canN device. Simply installing VeCanSetup won't do anything. 

With only USB CANbus devices that use Gs USB (not using slcan), you probably don't need VeCanSetup at all. The only reason would be to name the port for easier identification and since you only have one port that's probably unnecessary also.

I recommend installing VeCanSetup.

closed 2023-10-11 by kwindrem

bluey1, 2023-10-11:
Hi
Thank you for the prompt reply.
I agree the CANbus has not been identified.
I watched the installation by Andy on Offgrid Garage, but must have missed something.
I will do some more reading,  and see how I can identify the CAN device.
Is there a link/video  to help me identify the CAN device?
I did notice in the video by Andy, that the installation is slightly different, and he had more devices to select from.
Could it be that the USB device I am using is not supported?
Thank you

bluey1, 2023-10-11:
Hi
I have rerun  VECANsetup.
The Can1 port appears to be working with data packets updating.
But I still have no data on display page.
I notice on Andy's setup, he got an overlay prompt. I do not get that prompt when setting up.
What can be wrong?
Than you

[image]
[image]
[image]

kwindrem, 2023-10-11:
I don't know when that video was created but sounds like it is out of date. Victron made some changes to allow automatic configuration of CANbus interfaces and that necessitated a significant change to VeCanSetup.

To add an interface to what VeCanSetup will configure, you now need to select install, then add, then select a type of interface. If the USB interface is not specifically specified in the list, choose custom USB then fill in the necessary information.

VeCanSetup scans the system for potential interfaces and list them, or will ask for you to provide all the information for your specific interface. I suspect the latter would only be necessary if the interface was not plugged in when you run setup.

GS USB interfaces are automatically configured by Venus OS but they do not get a custom name and the order they appear depends on the order of discovery, so if you move one to a different USB port, they may swap their association with can0, can1 etc.
Raspberry PI CANbus hats and slcan interfaces are not recognized by Venus OS so configuring them via VeCanSetup is required (or you can create the needed configuration manually (dev rules, DT overlays, etc.)

I don't provide any guidance other than the ReadMe file.

kwindrem, 2023-10-11:
Unless there is a device such as a battery monitor, MPPT, etc on the CANbus connected to this interface, there will be nothing to display on the follow overview.

You may need to configure the port in the Series menu to match a device connected to that port. For example, if you have a BMS, you need to specify that in the CAN-but profile.