kwindrem/VeCanSetup #31: Seplos / EEL RS485 to USB Cable

maxpd1, 2023-11-08:
Hi,

according to the comment 25th of July the cable, which was part of the EEL 48V DIY battery case, should be usable: https://community.victronenergy.com/questions/196594/seplos-bms-mars-battery-case-wie-mit-raspberry-ven.html

But do I miss a step?

I installed all required packages:

[image]

I tried to install 2 different drivers for the adapter:

`root@raspberrypi2:~# /data/VeCanSetup/setup
--- starting setup script v3.6
resizing /dev/mmcblk0p2
resize2fs 1.45.7 (28-Jan-2021)
The filesystem is already 2566717 (4k) blocks long.  Nothing to do!

VeCanSetup adds CANbus ports to the Venus OS system
Raspberry PI includes 0 CANbus ports
an unlimited number of USB ports may be configured
one single or dual CANbus hat may be configured

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: i

CANbus interface definitions:
interface: ( can11 ) PCAN-USB or Victron CANUSB
  pcanusb (can11) 0x72:0101 123

interface: ( can12 ) CANable USB with Candelight firmware
  canable (can12) 1d50:606f 456

Available actions:
  Add a USB interface (a <interface>)
  Delete existing interface (d <interface>)
  blank line if done with changes

interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above: a 11

GS USB device types
  1) CANable USB with Candelight firmware
  2) PCAN-USB or Victron CANUSB
  3) custom GS USB
slcan device types -- not recommended
  4) CANable USB with slcan firmware or CANtact
  5) VScom USB-CAN+ slcan
  6) custom sclan USB

Choose a device type for can11 from the list above (by number) (cr for no change                                                                                                             ): 3

creating custom device

usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
select this device for can11 (y/n)?: y
enter optional name for can11 (cr for none): seplosusb
enter serial number manually (cr for none): 1234
adding serial number 1234 to can11 udev rules
can11 configuration updated

CANbus interface definitions:
interface: ( can11 ) custom GS USB
  seplosusb (can11) 0424:9514 1234

interface: ( can12 ) CANable USB with Candelight firmware
  canable (can12) 1d50:606f 456

Available actions:
  Add a USB interface (a <interface>)
  Delete existing interface (d <interface>)
  blank line if done with changes

interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above: a 12

GS USB device types
  1) CANable USB with Candelight firmware
  2) PCAN-USB or Victron CANUSB
  3) custom GS USB
slcan device types -- not recommended
  4) CANable USB with slcan firmware or CANtact
  5) VScom USB-CAN+ slcan
  6) custom sclan USB

Choose a device type for can12 from the list above (by number) (cr for no change                                                                                                             ): 1
enter optional name for can12 (cr for none): canable
enter serial number manually (cr for none): 234
adding serial number 234 to can12 udev rules
can12 configuration updated

CANbus interface definitions:
interface: ( can11 ) custom GS USB
  seplosusb (can11) 0424:9514 1234

interface: ( can12 ) CANable USB with Candelight firmware
  canable (can12) 1d50:606f 234

Available actions:
  Add a USB interface (a <interface>)
  Delete existing interface (d <interface>)
  blank line if done with changes

interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above: a 13

GS USB device types
  1) CANable USB with Candelight firmware
  2) PCAN-USB or Victron CANUSB
  3) custom GS USB
slcan device types -- not recommended
  4) CANable USB with slcan firmware or CANtact
  5) VScom USB-CAN+ slcan
  6) custom sclan USB

Choose a device type for can13 from the list above (by number) (cr for no change                                                                                                             ): 2
enter optional name for can13 (cr for none): pcan
enter serial number manually (cr for none): 345
adding serial number 345 to can13 udev rules
can13 configuration updated

CANbus interface definitions:
interface: ( can11 ) custom GS USB
  seplosusb (can11) 0424:9514 1234

interface: ( can12 ) CANable USB with Candelight firmware
  canable (can12) 1d50:606f 234

interface: ( can13 ) PCAN-USB or Victron CANUSB
  pcan (can13) 0x72:0101 345

Available actions:
  Add a USB interface (a <interface>)
  Delete existing interface (d <interface>)
  blank line if done with changes

interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above:
installing CANbus ports
removing scripts from old version of VeCanSetup
adding udev rules for can11
adding udev rules for can12
adding udev rules for can13
updating VeCanSetup udev rules

new interfaces will not become active until after a reboot
Reboot system now (y) or do it manually later (n): y
rebooting ...

Broadcast message from root@raspberrypi2 (pts/0) (Wed Nov  8 14:45:03 2023):
The system is going down for reboot NOW!
root@raspberrypi2:~#
`

But I don't see services or devices and the cable isn't flashling as it does when there is a successfull data transfer with a windows PC.

[image]

[image]

Regards 
maxpd

kwindrem, 2023-11-08:
If you add serial numbers and they aren't included in the information provided by the interface, they will NOT be activated. If the devices do not have a serial number then you can have at most ONE of that device type and you should not enter any serial number.

maxpd1, 2023-11-08:
I am not sure if the device has a serial number. At least I can't find any.
So I deleted the ports and installed it again without serial number. 
But I still don't have luck.

kwindrem, 2023-11-08:
By selecting a custom device type, the script will attempt to find devices that MIGHT be a CANbus interface and report that to you. If recognize the interface by what is provided, the script will build a configuration based on what is shown. If nothing is detected, you will need to provide some way for the dev rules to identify the device either by device id, vendor id, model number or serial number. You will need to get this information from the device's documentation (if provided) or by inspecting the port information reported in dmesg.

If you select a device that is supported by the script, most of the necessary information is filled in for you.

If you have multiple interfaces, some way of uniquely identifying each is essential.

It may be that the device you have is not actually a CANbus interface but actually requires a separate interface.

maxpd1, 2023-11-09:
sorry if I have to ask further questions.
how do I maintain the dev rules respectively provide the necessary information to the system?
was it last of your readme? if yes I am so sorry that I didn't find it.
in my log above lot of your started information were recognized by the system.

idVendor=0424, idProduct=9514, bcdDevice= 2.00

maxpd1, 2023-11-09:
ok, it got more cleare to me with dmesg

I repeated the process and took the following usb device with serial number and so on.
There is only one USB port used on the raspi.
But unfortunately I still get no date.
I am using the RS485 port instead the can (I tried both several times) as described in the user report mentioned in first post.

`root@raspberrypi2:~# /data/VeCanSetup/setup
--- starting setup script v3.6
resizing /dev/mmcblk0p2
resize2fs 1.45.7 (28-Jan-2021)
The filesystem is already 2566717 (4k) blocks long.  Nothing to do!

VeCanSetup adds CANbus ports to the Venus OS system
Raspberry PI includes 0 CANbus ports
an unlimited number of USB ports may be configured
one single or dual CANbus hat may be configured

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: i

no CANbus definitions found
Available actions:
  Add a USB interface (a <interface>)
  blank line if done with changes

interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above: a 10

GS USB device types
  1) CANable USB with Candelight firmware
  2) PCAN-USB or Victron CANUSB
  3) custom GS USB
slcan device types -- not recommended
  4) CANable USB with slcan firmware or CANtact
  5) VScom USB-CAN+ slcan
  6) custom sclan USB

Choose a device type for can10 from the list above (by number) (cr for no change): 3

creating custom device

usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
select this device for can10 (y/n)?: n

usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
select this device for can10 (y/n)?: n

usb 1-1.3: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
usb 1-1.3: Product: FT232R USB UART
usb 1-1.3: Manufacturer: FTDI
usb 1-1.3: SerialNumber: AB8A1KA2
usb 1-1.3: Detected FT232RL
usb 1-1.3: FTDI USB Serial Device converter now attached to ttyUSB0
select this device for can10 (y/n)?: y
enter optional name for can10 (cr for none):

Use serial number supplied above (y/n)?: y
adding serial number AB8A1KA2 to can10 udev rules
can10 configuration updated

CANbus interface definitions:
interface: ( can10 ) custom GS USB
  0403:6001 AB8A1KA2

Available actions:
  Add a USB interface (a <interface>)
  Delete existing interface (d <interface>)
  blank line if done with changes

interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above:
installing CANbus ports
removing scripts from old version of VeCanSetup
adding udev rules for can10
updating VeCanSetup udev rules

new interfaces will not become active until after a reboot
Reboot system now (y) or do it manually later (n): y
rebooting ...

Broadcast message from root@raspberrypi2 (pts/0) (Thu Nov  9 16:36:19 2023):
The system is going down for reboot NOW!
root@raspberrypi2:~#
`

kwindrem, 2023-11-10:
I believe your issue is that the BMS is not providing a CANbus equivalent over the USB cable you are using. If you have Cerbo, it has a full function CANbus port as well as one dedicated for use with a BMS. As others have described in the article you linked at the begging of this thread, this needs to plug into the CANbus port on your BMS/battery case with the appropriate cable -- all mentioned in that other thread.

For other platforms you may need to purchase a CANbus interface such as the Canable one I used for a while when running Venus OS on a Raspberry PI. It plugs into a USB port on the GX device and then with a home-made cable into the BMS CANbus port.

maxpd1, 2023-11-10:
ok, maybe I should buy a canable. I just wonder why the last comment in the linked topic uses successfully same cable with same BMS and same Venus os on raspi

maxpd1, 2023-11-11:
only to be sure. can you link the canable you used successfully?
btw. I saw in notification that you left a comment in the victron forum in the linked topic. but your comment isn't visible.

kwindrem, 2023-11-11:
The one I used was the CANable Pro but is currently showing as sold out.

There is a 2.0 version that is not isolated like the Pro is.

https://canable.io

To be stable, you need to reflash with Candlelight firmware. Candlelight is GS USB.

slcan software will sometimes work but Victron reported dropped packats with it.

There are also several USB CANbus interfaces listed at Victron that should work:

https://github.com/victronenergy/venus/wiki/RaspberryPi-CAN-Interfaces