kwindrem/VeCanSetup #32: can0 config is old format -- can't install

andyalford, 2023-11-25:
run again from command line, delete can0 and add it back to update config

-> I keep getting this message for can0 and can1, and at the same time, I do not seem to reliably get my can config to work.
-> For example, I add can10 and then restart and get can0 listed.

I think I have some old files stuck in my Pi system, can you advise what I should do?

Thanks

kwindrem, 2023-11-25:
It sounds like you might have a CANbus hat board plugged in. These will automatically configure as can0 and can1 if the dt overlay is in place in /u-boot/overlays and called from /u-boot/config.txt.

VeCanSetup will never create a can0 or can1 device because there is a conflict with what Venus OS configures automatically. CANbus hats will be configured as can8 and can9 and USB CANbus interfaces will be configured starting with can10.

Don't confuse names you might give the device with the actual device number. The name can be anything you want it to be but I've seen some use "can0" so this can get very confusing.

If you want to completely remove all the stored configurations, install the package from the setup script then delete /data/setupOptions/VeCanSetup directory and all its contents. It is important to uninstall first to make sure the active files the script installs are removed.

andyalford, 2023-11-27:
Thanks for the quick reply.  

I don’t have a CANbus hat board, but my installation has seen a few changes and gone through a few upgrades where I have used can0 / can1 a few times, so a cleanup seems to be a good idea.

 

I think a typo in your last para, but this is my plan…

*	uninstall the package from the setup script
*	then delete /data/setupOptions/VeCanSetup directory and all its contents
*	A reboot, etc
*	Then fresh install and use an unambiguous name like ‘GobelPower01’ 

 

I’ll write to confirm anything I learn, in case anyone else can benefit.

 

Thanks again for an excellent piece of work!!