kwindrem/VeCanSetup #33: No devices show after upgrade from V3.4 to V3.7

seamaster101, 2023-12-13:
Hi I have Isolated Waveshare hat with 2 can interfaces. it was forking fine before with version 3.4 installed.
the interfaces were showing as can8 and can9. 
Before the upgrade the devices were showing on can9 and all was working as expected - I could see NMEA2000 my tank levels.
After the upgrade, all devices start showing up on the can8 interface, but I cant see anymore more tank levels on the Venus OS.
Any idea what is the issue here? it is proven that the interface worked for months before without any issues. the updated appear to switch the names of both interfaces, and maybe that is creating an issue?
I even tried to disable can9 (the one that does not have devices now)  and leave active only can8 one with the devices, but that did not help either. Maybe Venus OS is expecting the read the device data from can9 as they were there before the upgrade. Please help.

kwindrem, 2023-12-13:
The current version of VeCanSetup is v3.7. You might try that, or maybe that is what you are running now.

You can roll back to the working version of VeCanSetup by setting the branch/tag in PackageManager to v3.4.

I have heard reports that ports on a CANbus hat board sometimes swap which device they get assigned to, but VeCanSetup should be nailing these down so this does not happen.

CANbus configurations are stored in settings under their canX assignments, so of the ports swapped, they will get configurations for the other port.

seamaster101, 2023-12-14:
I downgraded to V3.4 and it is same again:
I can see all my nmea2000 devices listed under services -> one(can8), but the data from these devices is not being displayed any longer e.g. tank levels. They were always displayed before. 
what should I do?

kwindrem, 2023-12-14:
Try deleting the port configurations, then uninstall thenreinstall VeCanSetup. Then add your configurations back in. I don't have much to go on here. I've played with a CAN hat but never actually used one. I used a USB CANbus interface for a while but my working system is on Cerbo now.

The main log of interest is the command dmesg which outputs info about the boot up process which is where port configurations are done. There is also a way to see what udev does. That's the mechanism to guide peripheral attachments.

There is a limited about of port configuration in the Services menu. You should check there to insure you have enabled the appropriate profile for generic N2K devices.

You might also try updating firmware to an older version. These can be found here:

http://updates.victronenergy.com/feeds/venus/

Placing the appropriate .swu file on a USB stick and inserting it into our PI and rebooting will switch to that firmware version.

Last resort is to flash a new SD card and reconfigure your system from scratch.

Sorry I can't be of more help but it does take some digging into the system.