kwindrem/VeCanSetup #35: Waveshare_2-channel_FD Overlay wrong?

Didi1965, 2024-01-25:
Hi, 
i have CAN-Hat Board MCP2518FD installed, but with the existing overlay no Interface dedected. 

With this parameters in u-boot/config.txt i get function:
dtparam=spi=on
dtoverlay=spi1-3cs
dtoverlay=mcp251xfd,spi0-0,interrupt=25
dtoverlay=mcp251xfd,spi1-0,interrupt=24

look at discription: https://www.waveshare.com/wiki/2-CH_CAN_FD_HAT 

Please can you check and modify the Overlay ?
https://github.com/kwindrem/VeCanSetup/tree/main/templatesRpiHat/Waveshare_2-channel_FD

thanks
BR Didi

kwindrem, 2024-01-26:
This appears to be a new revision of the board that requires different setup. I'll need to build a new template for it.

closed 2024-01-26 by kwindrem

dersch81, 2024-02-20:
Still wrong please use the correct overlay as written in the Waveshare Wiki and posted here by @Didi1965 

dtparam=spi=on
dtoverlay=spi1-3cs
dtoverlay=mcp251xfd,spi0-0,interrupt=25
dtoverlay=mcp251xfd,spi1-0,interrupt=24

then the 2 CH FD HAT works