kwindrem/VeCanSetup #36: Is MCP2518FD supported?

dersch81, 2024-02-20:
Hi i'm wondering if MCP2518FD is supported? 

https://www.waveshare.com/wiki/2-CH_CAN_FD_HAT

I would like to use that HAT on RPi4 because i can power the Pi with DC 8-26v over that expansion, too. This is very comfortable.

kwindrem, 2024-02-20:
It's hard for me to keep track of all the variation of these hat boards. I do support a 2-channel FD had that you could try.

dersch81, 2024-02-20:
I have installed RpiGPIO with alternate gpio list and VeCanSetup but this happens:

```
[    4.349115] pinctrl-bcm2835 fe200000.gpio: pin gpio19 already requested by fe200000.gpio; cannot claim for fe215080.spi
[    4.360215] pinctrl-bcm2835 fe200000.gpio: error -EINVAL: pin-19 (fe215080.spi)
[    4.367703] pinctrl-bcm2835 fe200000.gpio: error -EINVAL: could not request pin 19 (gpio19) from group gpio19 on device pinctrl-bcm2711
[    4.380120] spi-bcm2835aux fe215080.spi: error -EINVAL: Error applying setting, reverse things back
```

kwindrem, 2024-02-20:
The standard and alternate gpio_list will not work in all cases.
You need to manually sort out GPIO pin conflicts.

closed 2024-02-20 by kwindrem