kwindrem/VeCanSetup #38: Waveshare CAN 2CH FD HAT - Overlay spi1-3cs conflict with RpiGpioSetup GPIO16

dersch81, 2024-02-23:
As written at the RpiGpioSetup issue we need to replace spi1-3cs with spi1-1cs:

```
#### begin CAN overlay
# Waveshare 2-channel Isolated CANbus FD Hat
# instantiates as can0 and can1 ONLY !!!
# use udev rules to rename
[all]
dtoverlay=spi1-1cs
dtoverlay=mcp251xfd,spi0-0,interrupt=25
dtoverlay=mcp251xfd,spi1-0,interrupt=24
#### end CAN overlay
```

You already deliver this overlay at /data/VeCanSetup/spi1-1cs.dtbo

This file has to move to /u-boot/overays as well.

dersch81, 2024-02-23:
As a side note. spi1-3cs is only neccessary in case of two 2CH FD CAN Hats in a stack. So nearly nobody will use that on the Venus side, i guess. 

That's why we can go in general with the spi1-1cs setting. In case of a stack there will not be any gpio free to use with RpiGpio and this will be a special individual case.

dersch81, 2024-03-03:
Hi, can you merge this request?

kwindrem, 2024-03-03:
Sorry, thought I'd done this already, but not. I'll get on it.

dersch81, 2024-03-04:
Many thanks. BTW: i just noticed that the FD HAT is also working with that entry in u-boot/config.txt without VeCanSetup installed out of the box ;)

kwindrem, 2024-03-04:
Yes, I don't doubt that it works without VeCanSetup. There will of course be conflicts with using GPIO pins for digital inputs and relay outputs set up in RpiGpioSetup and the spi1-3cs that's included in the Venus OS image.

The other issue is some CANbus hats randomly switch channels on can0 and can1 even though those are specified in confit.txt. And there would be conflicts with any USB CANbus adapters installed at the same time. VeCanSetup addresses these issues.

dersch81, 2024-03-04:
But the conflicts with RpiGpio are adressed from the RpiGpio side with specific mapping (on what we worked together)?

What we also found is, that for this specific HAT in combination with RpiGpio the spi1-1cs is the right choice to use. 

Anyway. Many thanks for your support. Now we have a full functional 2CH FD HAT without any issue with RpiGpio AND ShutdownPin anymore. We can use now everything together without any compromise.

kwindrem, 2024-03-04:
Are you saying that if you specify spi1-1cs in config.txt that spi1-3cs.dtbo doesn't enable conflicting pins? And that the hat works without RpiGpio conflicts?

If so, do I need to install spi1-1cs.dtbo? This would be good if so.

dersch81, 2024-03-04:
Here is the PIN List of the HAT for your reference

[image]