kwindrem/VeCanSetup #39: no CANbus

fitschinet, 2024-04-18:
now i will install a new Venus os with a waveshare can hat Rev2.1
i cant install VeCanSetup because i got all time an error

VeCanSetup adds CANbus ports to the Venus OS system
Raspberry PI includes 0 CANbus ports
an unlimited number of USB ports may be configured
one single or dual CANbus hat may be configured

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display setup log (s) outputs the last 100 lines of the log

Choose an action from the list above: i

no CANbus definitions found
Available actions:
  Add a USB interface (a <interface>)
  blank line if done with changes

interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above:

what can i do?

kwindrem, 2024-04-18:
You need to add an interface. In your case select hat then select the hat type and answer additional questions presented.

closed 2024-04-18 by kwindrem

fitschinet, 2024-04-19:
[image]

kwindrem, 2024-04-19:
The instructions are right above the prompt:

add

then

hat

or add hat