kwindrem/VeCanSetup #42: invalid key/value pair in file /etc/udev/rules.d/VeCanSetup.rules on line 14, starting at character 32 (',')

mvincm, 2024-09-07:
Hello,

Can is working but every reboot I notice this line in logs (dmesg). Is something wrong with the script or I have something messed up with my specific configuration?

Any ideas?

Thanks in advance.
Mat

kwindrem, 2024-09-07:
Without seeing the udev rules file which is generated automatically during setup I can't tell what might be wrong.

mvincm, 2024-09-09:
Hello,

Yes, you are totally right. Here are log files and file. Should I search some else more? But still can-us adapter (canable) works fine for VenusOS.

dmesg:
[    5.046980] hub 1-1.3:1.0: 7 ports detected
[    5.103059] udevd[168]: invalid key/value pair in file /etc/udev/rules.d/VeCanSetup.rules on line 14, starting at character 32 (',')
[    5.130958] udevd[169]: starting eudev-3.2.9

rules:
root@raspberrypi2:~# cat /etc/udev/rules.d/VeCanSetup.rules
# udev rules for gs USB devices

# replace ...PH (place holder) with actual values
# DELETE the placeholder lines if they aren't used !!
# rename file to e.g., can0.rules
#

ACTION=="add", SUBSYSTEM=="net",\
  ENV{ID_VENDOR_ID}==0x72",\
  ENV{ID_MODEL_ID}=="0101",\
  ATTRS{serial}=="HQ2238GMMAF",\
  NAME="can10",\
  ENV{VE_NAME}="can10PCAN (can10)",\
  RUN+="/usr/local/bin/startGsUsbCan.sh can10"
  RUN+="/usr/bin/logger [udev] CANbus device added:\
    can10PCAN (can10)\
    can10\
    VENDOR_PH\
    MODEL_PH\
    HQ2238GMMAF\
    "
root@raspberrypi2:~#

mvincm, 2024-09-10:
Thanks mate! A will check it tomorrow evening. Once again thank you for your support.