kwindrem/VeCanSetup #43: Waveshare RS232 RS485 CAN Board supported?

JiMcNulty, 2024-09-09:
It's a pretty cool device

[https://www.waveshare.com/rs232-rs485-can-board.htm](https://www.waveshare.com/rs232-rs485-can-board.htm
)

Doesn't have anything new, just combining everything together, so I figure it should work but wanted to make sure before I buy it
[image]

kwindrem, 2024-09-09:
Not specifically supported but the CANbus portion might work if it uses the same pins as other Waveshare CANbus boards.

closed 2024-09-09 by kwindrem

JiMcNulty, 2024-11-22:
So I just got around to it
Had to manually set it with this

`dtparam=spi=on
dtoverlay=i2c0 
dtoverlay=spi1-3cs
dtoverlay=sc16is752-spi1,int_pin=25
dtoverlay=mcp2515,spi0-0,oscillator=16000000,interrupt=23
dtoverlay=mcp251xfd,spi0-1,interrupt=24`

Had a problem that only 1 can was initialized (commented out one and the other one came up after restart)
Used VeCanSetup with one of the hat, and it fixed it somehow, still working after uninstalling VeCanSetup

Wanted to ask if you have an idea why?
(Connecting my Ve.can mppt still doesn't work, all TX packets are dropped and have some really funky kernel errors)
`
[  616.687754] mcp251xfd spi0.1 can1: bus-off, scheduling restart in 100 ms
[  616.802787] IPv6: ADDRCONF(NETDEV_CHANGE): can1: link becomes ready
`

JiMcNulty, 2024-11-22:
I figured it out, each interface needs to be in a different section 
Surprising but that's how it's working now

`root@raspberrypi4:~# cat /u-boot/config.txt
enable_uart=1
kernel=u-boot.bin

[all]
device_tree=bcm2711-rpi-4-b.dtb
dtparam=spi=on
dtoverlay=i2c0
dtoverlay=spi1-3cs
dtoverlay=sc16is752-spi1,int_pin=25
dtoverlay=mcp2515,spi0-0,oscillator=16000000,interrupt=23
dtoverlay=mcp251xfd,spi0-1,interrupt=24

[all]
dtparam=spi=on
dtoverlay=mcp2515-can0,spi0-0,oscillator=16000000,interrupt=23

`

kwindrem, 2024-11-22:
VeCanSetup makes file system changes when you run the setup script. If you uninstall the package without first running the uninstall option then the file system changes remain.

Not all hats work with Venus OS for various reasons. There are often also GPIO conflicts that need to be understood and worked around. It's not possible for me to accommodate all GPIO configurations so I'm afraid you are on your own.

Glad you got it working.

With recent changes to Venus OS, Victron and I both recommend using USB CANbus interfaces that use the GS Lib interface. These will automatically configure without running VeCanSetup for configuration but the order multiple interfaces appears is based on the order they are discovered. Moving one to a different USB port is enough to change that order. So I recommend using VeCanSeup and specifying enough (serial numbers for example) to uniquely identify each interface. With that done, the ports will remain consitent.

JiMcNulty, 2024-11-22:
Thanks for the detailed reply!

(I only did the uninstall option without uninstalling the package)

I'm already fully invested with this uber ultra multi can interface

Wanted to contribute a PR to add this hat, which I believe will be very popular, but you say that you can Victron will push for USB so I'm not sure now it will be worth the hassle