kwindrem/VeCanSetup #44: Choose action from list above: hat invalid action, please try again

Nvanderbeek, 2024-09-15:
when I run command line to install the CAN port connected to GPI 8,9,10,11,25 on my waveshare https://www.waveshare.com/wiki/CM4-IO-WIRELESS-BASE-B

I have selected hat as option but run into a error:
interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above: hat
invalid action, please try again

no CANbus definitions found
Available actions:
  Add a USB interface (a <interface>)
  blank line if done with changes

kwindrem, 2024-09-15:
Is this working or is something still wrong?

I don't recall how I check for CANbus hats but this would usually result in the script reporting the hat interface. As you say it could be because you manually entered the lines in config.txt.

Nvanderbeek, 2024-09-16:
Yes, it is working. After removing the line

closed 2024-09-16 by kwindrem