kwindrem/VeCanSetup #45: Waveshare USB-CAN-A Challenges

ScottS45, 2024-09-21:
Hi,

First I would like to thank you for all your efforts for the community.

I have a Waveshare USB-Can-A that I have attached to a Xantrex XC Pro supporting RV-C.  I have the latest version of VanCanSetup, as well as Helper, Gui.  I have attempted to install via the command line.

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display log (l) outputs the last 100 lines of the log

Choose an action from the list above: i

CANbus interface definitions:
interface: ( can10 ) custom GS USB
  Xantrex XC Pro (can10) 1a86:7523
  
  Choose a device type for can10 from the list above (by number) (cr for no change): 3

creating custom device

usb 1-1.1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
usb 1-1.1: Product: USB Serial
select this device for can10 (y/n)?: y
enter optional name for can10 (cr for none): Xantrex XC Pro
enter serial number manually (cr for none):
can10 configuration updated

CANbus interface definitions:
interface: ( can10 ) custom GS USB
  Xantrex XC Pro (can10) 1a86:7523

Available actions:
  Add a USB interface (a <interface>)
  Delete existing interface (d <interface>)
  blank line if done with changes

interface may optionally be added to the action, (e.g. a hat or d 11)
USB interface must be 10 or greater
Hat interface should be entered as 'h' or 'hat'

Choose action from list above:
installing CANbus ports
removing scripts from old version of VeCanSetup
adding udev rules for can10
updating VeCanSetup udev rules
installing spi1-1cs overlay

new interfaces will not become active until after a reboot
Reboot system now (y) or do it manually later (n): y
rebooting ...

ScottS45, 2024-09-21:
After the reboot I 

root@raspberrypi4:~# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,DYNAMIC,UP> mtu 1450 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether d8:3a:dd:2d:96:d8 brd ff:ff:ff:ff:ff:ff
3: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
4: wlan0: <BROADCAST,MULTICAST,DYNAMIC,UP,LOWER_UP> mtu 1450 qdisc pfifo_fast state UP mode DORMANT group default qlen 1000
    link/ether d8:3a:dd:2d:96:d9 brd ff:ff:ff:ff:ff:ff
5: ll-eth0@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1450 qdisc noqueue state LOWERLAYERDOWN mode DEFAULT group default qlen 1000
    link/ether 92:ee:62:39:8b:d8 brd ff:ff:ff:ff:ff:ff

I do not see Can 10 in GUI, I assume it should show up under Services.  

**root@raspberrypi4:/etc/udev# cat rules.d/VeCanSetup.rules**

# udev rules for gs USB devices

# replace ...PH (place holder) with actual values
# DELETE the placeholder lines if they aren't used !!
# rename file to e.g., can0.rules
#

ACTION=="add", SUBSYSTEM=="net",\
  ENV{ID_VENDOR_ID}=="1a86",\
  ENV{ID_MODEL_ID}=="7523",\
  NAME="can10",\
  ENV{VE_NAME}="Xantrex XC Pro (can10)",\
  RUN+="/usr/local/bin/startGsUsbCan.sh can10"
  RUN+="/usr/bin/logger [udev] CANbus device added:\
    Xantrex XC Pro (can10)\
    can10\
    1a86\
    7523\
    "

      SetupHelper now logged to /var/log/PackageManager/current
**root@raspberrypi4:/etc/udev# cat /var/log/PackageManager/current**
@4000000066eb623f35222e34 WARNING:>>>> PackageManager v8.20 starting
@4000000066eb62491f6dbcfc WARNING:starting boot-time reinstall
@4000000066eb62501e5a06f4 WARNING:boot-time reinstall complete
@4000000066eb625021561c6c WARNING:checking SetupHelper
@4000000066eb62512a9f04b4 SetupHelper/setup: --- starting setup script v8.20 action: CHECK
@4000000066eb625722c4c79c WARNING:checking GuiMods
@4000000066eb6258086cb724 GuiMods/setup: --- starting setup script v10.66 action: CHECK
@4000000066eb625b212f75e4 GuiMods/setup: WARNING: styles.css no active file for patch - skipping
@4000000066eb625b2da8e184 WARNING:checking ShutdownMonitor
@4000000066eb625c18222f3c ShutdownMonitor/setup: --- starting setup script v3.6 action: CHECK
@4000000066eb625c35efc31c WARNING:checking RpiGpioSetup
@4000000066eb625d176dcf74 RpiGpioSetup/setup: --- starting setup script v4.10 action: CHECK
@4000000066eb625d38930e1c WARNING:checking RpiDisplaySetup
@4000000066eb625e1ba3b8c4 RpiDisplaySetup/setup: --- starting setup script v3.13 action: CHECK
@4000000066eb625e2d02b674 WARNING:checking VeCanSetup
@4000000066eb625f0c34b3d4 VeCanSetup/setup: --- starting setup script v3.14 action: CHECK
@4000000066eb626001c4c81c WARNING:checking RpiTemperature
@4000000066eb626021d18a14 RpiTemperature/setup: --- starting setup script v1.2 action: CHECK
@4000000066eb63340c8836bc VeCanSetup/setup: --- starting setup script v3.14 action: NONE
@4000000066eb64602c68a8f4 VeCanSetup/setup: --- starting setup script v3.14 action: NONE
@4000000066eb64bd23ec4bf4 VeCanSetup/setup: can10 configuration updated
@4000000066eb64cc39d2c9ac VeCanSetup/setup: installing CANbus ports
@4000000066eb64cd017b99bc VeCanSetup/setup: removing scripts from old version of VeCanSetup
@4000000066eb64cd06a70a84 VeCanSetup/setup: adding udev rules for can10
@4000000066eb64cd07b24c7c VeCanSetup/setup: updating VeCanSetup udev rules
@4000000066eb64cd0da2de1c VeCanSetup/setup: installing spi1-1cs overlay
@4000000066eb64f636c13834 VeCanSetup/setup: rebooting ...
@4000000066eb64ff321138fc SetupHelper/reinstallMods: reinstallMods starting
@4000000066eb64ff3528d16c SetupHelper/reinstallMods: reinstallMods finished
@4000000066eb650122ae74c4 *** CCGX booted (0) ***
[... 127 lines cut ...]
@4000000066ef12032573b32c VeCanSetup/setup: adding udev rules for can10
@4000000066ef120326a64b74 VeCanSetup/setup: updating VeCanSetup udev rules
@4000000066ef12032c5f8904 VeCanSetup/setup: installing spi1-1cs overlay
@4000000066ef122828e3b854 VeCanSetup/setup: rebooting ...
@4000000066ef1230329011a4 SetupHelper/reinstallMods: reinstallMods starting
@4000000066ef123035bd510c SetupHelper/reinstallMods: reinstallMods finished
@4000000066ef12322286a944 *** CCGX booted (0) ***
@4000000066ef123b0298b21c WARNING:>>>> PackageManager v8.20 starting
@4000000066ef124506fe88f4 WARNING:starting boot-time reinstall
@4000000066ef124c086237a4 WARNING:boot-time reinstall complete
@4000000066ef124c0bcd0cac WARNING:checking SetupHelper
@4000000066ef124c31c9431c SetupHelper/setup: --- starting setup script v8.20 action: CHECK
@4000000066ef12530ce35eac WARNING:checking GuiMods
@4000000066ef1253311bf3ec GuiMods/setup: --- starting setup script v10.66 action: CHECK
@4000000066ef1256344e8a84 GuiMods/setup: WARNING: styles.css no active file for patch - skipping
@4000000066ef12570b99a484 WARNING:checking ShutdownMonitor
@4000000066ef125732d41f84 ShutdownMonitor/setup: --- starting setup script v3.6 action: CHECK
@4000000066ef1258163fbf54 WARNING:checking RpiGpioSetup
@4000000066ef1259010c8054 RpiGpioSetup/setup: --- starting setup script v4.10 action: CHECK
@4000000066ef12591ce9e0dc WARNING:checking RpiDisplaySetup
@4000000066ef125a05f9fdbc RpiDisplaySetup/setup: --- starting setup script v3.13 action: CHECK
@4000000066ef125a162b396c WARNING:checking VeCanSetup
@4000000066ef125a31a37b3c VeCanSetup/setup: --- starting setup script v3.14 action: CHECK
@4000000066ef125b060d04d4 WARNING:checking RpiTemperature
@4000000066ef125b20afc27c RpiTemperature/setup: --- starting setup script v1.2 action: CHECK

kwindrem, 2024-09-22:
There are two common interface protocols for CANbus USB interfaces. The preferred is GS USB and works well with Victron firmware. The other is slcan. Victron has reported issues with dropped packets with slcan interfaces and they are not recommended but work OK in situations where devices send regular updates making a dropped packet less problematic.

It appears that the Waveshare USB-Can-A device uses neither of these protocols. That is most likely why it is not showing up in the services menu.

ScottS45, 2024-09-22:
Do you have a suggestion(s) on CAN to USB adapter that supports either of those??

kwindrem, 2024-09-22:
I have only tested the CANable device.
https://canable.io
This comes with slcan firmware installed but can be easily refreshed to GS USB (Candlelight)

Other devices are listed on the Victron web site. Sorry, don't have a current link because things have moved around since I last looked.

ScottS45, 2024-09-22:
Thanks for your assistance!

ScottS45, 2024-09-29:
I picked up a CANable (https://www.amazon.ca/dp/B0CRB8KXWL) unit with candlelight and it shows up under ip link and as a service in VenusOS.  No info is showing in the GUI yet.  Though under network it only shows 3 packets for TX vs 29 for RX.   State is Error-Passive.  That has me wonder if the work mode needs to changed?

kwindrem, 2024-09-30:
That depends on what is connected to the device. I'd say of you are seeing it in the services menu then you are good to go. The error-Passive is the normal (non error state).

ScottS45, 2024-09-30:
It is a Xantrex XC 3000 Inverter/Charger.  It was on, though the status in the GUI did not reflect that or when it was pulling a load.  That has me thinking I am missing something still.

kwindrem, 2024-09-30:
I do not think Xantrex inverter/chargers will show in the Victron GX device.

ScottS45, 2024-10-07:
> I do not think Xantrex inverter/chargers will show in the Victron GX device.

Are you aware of someone or a repo, that has done a driver for a device over RV-C ?  I am looking for something I can have a look at and see if I can modify it for the Xantrex.  I have found a doc from them for the DN names supported.

kwindrem, 2024-10-07:
Sorry, I don't know of any third party mods to support Xantrex inverter/chargers.

ScottS45, 2024-10-07:
It does not need to be for them, but for any device.   I am a software dev, and was going to look at creating my own code to support it in VenusOS.  Just was looking for a starting point.

h-c-c, 2025-06-03:
> Are you aware of someone or a repo, that has done a driver for a device over RV-C ? I am looking for something I can have a look at and see if I can modify it for the Xantrex. I have found a doc [...]

@ScottS45 did you ever find any solution?

ScottS45, 2025-06-03:
> I do not think Xantrex inverter/chargers will show in the Victron GX device.

Are you aware of someone or a repo, that has done a driver for a device over RV-C ? I am looking for something I can have a look at and see if I can modify it for the Xantrex. I have found a doc from them for the DN names supported.

@ScottS45 did you ever find any solution?

I have not found a pre-coded solution, no.  I have been thinking about this again, the last couple of weeks actually.  I found hardware that I connect to my raspberry Pi (or VE) to translate, and did up wiring to the Xantrex.  I see messages being sent, but the VenusOS doesn't know what to do.  I started looking at Python so I can perhaps write something but I took a break.  I was just starting to look at this again.   It is a bit of learning on my part to figure out what to do.  If I could find existing code for something else communicates over RV-C, that works, that would go a long way to speeding it up for me.

ScottS45, 2025-06-03:
Fyi, I picked up this to help with coms between the two.  Jhoinrch USB to CAN Bus Converter Base Open-Source Hardware CANable Supports Multiple Firmware and Operating Modes.   It seems to be better than the waveshare I first got.