kwindrem/VeCanSetup #47: Waveshare 2 Channel CAN HAT+ install issue

AkosT3, 2025-03-28:
Dear Kevin,
I plan to migrate my Venus Os from Cerbo GX MK2 to Raspberry Pi4 expanded by a 2CH Waveshare CAN HAT+ due to the issues with can2 on the Cerbo.
As mentioned, I bought a 2CH isolated Waveshare CAN HAT+, but having troubles installing it.
I've tried all possible combinations to make it work, but no CAN ports appear in Venus OS in Services tab.
As I understand, the HAT+ is something between the Non-FD HAT and FD HAT.
The pinout, based on info on Waveshare wiki is different from both other versions.

No matter, what I change in overlays and config, I get the same error message from dmesg |grep spi:
[    6.238614] mcp251x spi0.0: MCP251x didn't enter in conf mode after reset
[    6.245685] mcp251x spi0.0: Probe failed, err=110
[    6.250571] mcp251x: probe of spi0.0 failed with error -110
[    7.269965] mcp251x spi0.1: MCP251x didn't enter in conf mode after reset
[    7.277004] mcp251x spi0.1: Probe failed, err=110
[    7.281924] mcp251x: probe of spi0.1 failed with error -110

Could you please, upgrade the VeCanSetup, to include this model of HAT in the setup, please?
I hope, you shall have time to do the upgrade, so we, I suppose others struggle with the same issue as well, can install this type of hat 
with your VeCanSetup in the future.
Thank you in advance for your work and efforts.

kwindrem, 2025-03-28:
There are many GPIO conflicts between this board and relays and digital inputs. It would be possible to create a custom gpio_list and install it using RpiGpioSetup to move things around. You also need to change the commands to relocate the interrupt lines in config.txt.

Without a module to test with any changes I make would be hit and miss.

You are really best off using USB based CANbus interfaces as they are more easy to integrate into the system.

kwindrem, 2025-04-06:
I have released a version of VeCanSetup and RpiGpioSetup that supports the Waveshare Can Plus hat.

I tested both can interfaces using my SeeLevel tank sensor system.

You need to update both packages.

There is also an update to SetupHelper that fixes a problem where some files were not uninstalled properly. This package should be updated also.

closed 2025-04-06 by kwindrem

AkosT3, 2025-04-06:
Hello Kevin,

I just tried to update, but the new versions are not available on github, yet.

I shall try tomorrow.

 

I’d have a question>

1.	Should I uninstall the SetupHelper first and then install the new version, or 
2.	Is it enough to update in over the PackageManager?
3.	The same question with VeCanSetup, should I uninstall first and then make a new install, or just update in PackageManager?

 

Thank you in advance for guidance.

 

Have a nice day!

 

Regards,

 

Akos

kwindrem, 2025-04-06:
I have seen it take up to 30 minutes for the info cached somewhere in the internet (especially the package version which I retrieve independent of the entire package) to update. Maybe your location takes longer. One thing you can try is to select the package then download it manually. Sometimes that will fetch the newer version.

Just update the three packages from PackageManager. No need to uninstall first.

You will need to run /data/RpiGpioSetup/setup and /data/VeCanSetup/setup from the command line since this is the only way to change the configuration.

If you have already created a hat device in VeCanSetup, you will need to delete the hat configuration so you can add the new one.

Once configured, these packages will reinstall with each firmware update.

kwindrem, 2025-04-06:
My error. I failed to create releases for SetupHelper and VeCanSetup. Please check PackageManager in a few minutes and you should see v8.31 and v3.15, respectively.

AkosT3, 2025-04-08:
Hello Kevin,
I just installed the RpiGpioSetup and afterwards the VeCanSetup, and it looks, everything installed perfectly. The CAN interfaces appeared in the system tab and are configurable. 
Thank you for your support. Seems everything is working as it should.

kwindrem, 2025-04-08:
Thanks for the report. Glad it worked out.