kwindrem/VeCanSetup #48: VeCanSetup not installing

rrusek, 2025-07-13:
Hi,
I was hoping that you could help me finalise the installation of VeCanSetup.

I have a Victron Multi RS Solar hybrid connected to a Raspberry Pi running Venus OS and in the interim using a VE.Direct adapter.
VRM reports the Multi RS is not fully supported when using VE Direct and the error message I get is:
“Unsupported device found. This product is not supported when connected via VE.Direct. Connect via VE.Can instead.”

I want to use a PCAN USB adapter and see that VeCanSetup supports that interface  in the description.
Package Manager indicates that SetupHelper has been successfully installed and VeCanSetup has moved from Inactive packages to Active packages. 
However, the installation does not proceed past the download prompt and will not allow me to select and install the PCAN driver. 
The VeCanSetup Installed tab only shows two dashes for the ver #. 

Please see the attached screenshot. 

Any assistance would be appreciated.

Cheers
Richard

[image]

closed 2025-07-13 by kwindrem

kwindrem, 2025-07-27:
Your pinout appears correct according to Victron documentation. If your adapter has the Victron RJ-45 connectors, then it should be matter of plugging in a standard ethernet cable between the adapter and the Multi. If the adapter uses the DB-9 then I'd check carefully that you've made the correct connections at that end.

If can0 is showing up in Services then VeCanSetup will not help.

For PCAN, VeCanSetup is looking for devices that advertise a vendor ID of 0x72 and a model ID of 0101. If it finds one while running the setup script it should be shown with a prompt to use that device. If you look at dmesg you should see the adapter in that output.

If VeCanSetup does not find the device then I suspect the parameters it is searching for are not correct. I have not tested VeCanSetup with a PCAN device so there may be errors in my code.

If VeCanSetup finds a device and you continue with configuration, it builds udev rules to move the device to can11 or whatever port you associate with it. This is obviously not happening.

You might ask in the Victron Community forum for help with this specific interface.

kwindrem, 2025-07-28:
Cerbo is a significant investment but so are all the bits and pieces needed to assemble an official CANbus network.

 was running Venus OS on an RPI for a while with VE.Bus, several VE.Direct and CANbus interfaces, but moved to Cerbo when I had the chance. Cerbo is slower than the RPI 4 but as you say is plug and play.